LumaTeam/Luma3DS #664: An exception occurred Stack Dump

MrFrawgg, 2017-06-27:
I followed this guide https://3ds.guide/installing-boot9strap-(browser) exactly!

On an Nintendo 3DS XL firmware 8.1.

I get to Section IV - Configuring Luma3DS and select "Show NAND or user string in System Settings" and then press START to save. After that I get the crash dump in the picture.

I can get back into Luma by holding Select at boot or by deleting config.bin.

Other than that I'm stuck. Never had this issue before.

Crash Dump:
https://3.sharebylink.com/1/?YR0eDvrSSv8k

Hikari-chin, 2017-06-27:
Could you try a newer build, like [this one](https://astronautlevel2.github.io/Luma3DS/builds/Luma3DS-e67c6ed.zip)?

Margen67, 2017-06-27:
Do astronautlevel2's builds work properly?

MrFrawgg, 2017-06-27:
I figured it out. I needed to use a Legacy version of boot.firm because I was on 8.1 firmware.

The guide didn't state that until the next page a couple paragraphs down. After that I was golden.

Hopefully this will help anyone in the future.

TheNain38, 2017-06-27:
@MrFrawgg Well, you can close this issue then

closed 2017-06-27 by Hikari-chin

rj197, 2020-12-03:
how to fix the stack dump because i need help please

rj197, 2020-12-03:
on my 3ds