LumaTeam/Luma3DS #665: Data Abort on launching memory-extended titles on some O3DS consoles

hedgeberg, 2017-06-27:
**System model:**
Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

Sys 11.2.0-35U

**Entrypoint (How/what you're using to boot Luma3DS):**

B9S

**Luma3DS version:**

Luma 8.0 Release

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: (x)

Set developer UNITINFO: ( )

--

Hey guys.

Luma 8.0 on some O3DS units, for some reason, crashes during firmlaunch for Ever Oasis. Quick review of community via ReSwitched and Nintendo Homebrew discord servers implies this is an uncommon but not unique issue. Unsure if this is the case on older versions of Luma, or on updated firmwares, or even with other firmlaunched titles. Will be testing as time allows but for now figured it was worth listing an issue. 

Seems to be a flaw in Agora, but i have no idea what the issue could be past that. If any others are having issues, please ping me so I can collect info, im hedgeberg in the nh discord. I'd like to collect info without polluting the heck out of this thread. 

Also worth noting, potentially: this O3DS is known to be deviant from the standard market O3DS, as there is a clock pad exposed at the nand cluster, making clk accessible by routes other than lifting the cartslot and soldering to a via. This pad was removed at some unknown point, and no images of this type of O3DS have been taken. If others have this issue, can provide an image to improve debugging, so we can see if this is the source of the issue. It's unlikely, but any correlating info may be helpful. 

**Dump file:**

[crash_dump_00000003.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/1104193/crash_dump_00000003.dmp.zip)

hedgeberg, 2017-06-27:
Fwiw I am now installing Smash 4 on this console to test if it is all extended memory titles or just Ever Oasis. I will test a firm update next and then a luma rollback.

Hikari-chin, 2017-06-27:
I ran into this **exact** issue earlier. You're likely using the EUR version of Ever Oasis on a US console; to fix it, you must check "Enable game patching" and put a proper locale.txt (EUR EN) in /luma/titles/00040000001A4900.

closed 2017-06-27 by Hikari-chin

hedgeberg, 2017-06-27:
Appreciated. Posting link to NH discord so folks are aware.