LumaTeam/Luma3DS #666: An exception occurred when N3DS boot

david-zzj, 2017-06-27:
**System model: New 3DS

**SysNAND version (+emu/redNAND version if applicable): 9.2.0-20J SysNAND

**Entrypoint (How/what you're using to boot Luma3DS): Boot9Strap

**Luma3DS version: 8.0

**Luma3DS configuration/options:**
Enable game patching: (YES )
Show NAND or user string in System Settings: (YES )

**Explanation of the issue:**
I followed this guide https://3ds.guide/, On an N3DS firmware 9.2sys.
When i Configuring Luma3DS and  press START to save,  I get the crash dump screen.

i am very i can't upload the picture, i described in txt.

processor：ARM11 (core 3)
type：data abort
current process：qtm（0004013000004202）

david-zzj, 2017-06-27:
I feel it very strange that 0004013000004202 impossible exist in the n3ds ver 9.2.0-20J  titles list.
Qtm services in new3ds could be 0004013020004202.
I have try CTR 9.2J and CTR 11.2J, but the same crash dump screen.

HugaShire, 2017-06-27:
Read the wiki (https://github.com/AuroraWright/Luma3DS/wiki/FAQ-and-troubleshooting this one)

genehung2003, 2017-06-28:
I got the same issue on my region changed 3ds
It was originally a Japanese 3ds, then I change it into a USA console following plailcet's guide
For now, I found two games that will break, which are ever oasis and heroes of ruins
Wish there's a solution for this

BTW, I've seen plailcet saying that CTR transfer will break reboot patches
And formatting your system can solve this
I'll try formatting my console as soon as I get a card reader around and will report then.

PS: forgot to mention that my console is 11.4 with 1.2 b9s on luma 8.0

genehung2003, 2017-06-28:
Yo guys, I fixed this problem for heroes of ruin in no time
Just update your luma through luma updater to the latest hourly build, and the issue just vanished.

david-zzj, 2017-06-28:
that's a good news，i will try later.
THX，bro.

daedreth, 2017-06-28:
@genehung2003 who is this plailcet that you speak of so highly?

Hikari-chin, 2017-06-28:
I don't think this is a Luma issue; had two people test this on N3DS 9.2J and 11.4J, and they didn't have this problem. Could you try the [8.1 release](https://github.com/AuroraWright/Luma3DS/releases/latest) of Luma? If that doesn't work, could you try putting Luma's boot.firm on your CTRNAND with GodMode9 and booting without your SD card?

closed 2017-06-28 by Hikari-chin

david-zzj, 2017-08-11:
i have try the 8.1.1 of luma and the latest hourly build, didn't work, the same crash dump screen.
lol.............