LumaTeam/Luma3DS #668: ACNL crashes while using NTR in a friends town.

Derek1199, 2017-06-28:
**System model:**
N3DS
**SysNAND version (+emu/redNAND version if applicable)
11.4.0-37U 
**Entrypoint (How/what you're using to boot Luma3DS):**
Latest commit of b9s.
**Luma3DS version:**
https://github.com/AuroraWright/Luma3DS/commit/e67c6ed3ee6804d29aca5d5e4b09493de514ceba

*Luma3DS configuration/options:**

Default EmuNAND: (*)

Screen brightness: (1)

Splash: ( )

PIN lock: ( )

New 3DS CPU: (*)

--

Autoboot EmuNAND: (*)

Use EmuNAND FIRM if booting with R: (*)

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (*)

Show NAND or user string in System Settings: (*)

Show GBA boot screen in patched AGB_FIRM: (*)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
While using rydog's acnl NTR cheats, going to a friend's town and then pressing the "a" button causes an exception to occur. This has been happening since around latest commit or a few before. (I didn't play yesterday) I'm also using latest commit of the plugin but yeah this is an issue that needs fixed now.

**Steps to reproduce:**

1. Use latest commit of luma and the latest commut of the Acnl NTR plugin cheat by Rydog.
2. Go to a friend's town and press the a button. You will be greeted with an arm 11 exception.

**Dump file:**

DeathAngel74, 2017-06-28:
I'm running the latest commit of luma, b9s, with game patching turned on in luma config. No crashes. Maybe NTR is the problem, not luma cfw?

Derek1199, 2017-06-28:
Did you go to a friend's town and press "a"?

DeathAngel74, 2017-06-28:
Yep. This sounds more like an NTR problem than a LUMA3DS problem though.

Derek1199, 2017-06-28:
I see. Hmm. Try using latest commit of the NTR cheats by rydog. It's an NTR
problem, but didn't occur until using latest commit of luma. So idk.

Derek1199, 2017-06-28:
I did compile luma myself though.

DeathAngel74, 2017-06-28:
One way to check....try friend's town, pressing "a", without NTR and with plugins disabled. Then you'll know for sure if its NTR or Luma.

Derek1199, 2017-06-28:
It's NTR though. But it didn't occur until using latest luma commit. Weird.

DeathAngel74, 2017-06-28:
You may have to revert a few commits of NTR to find one that doesn't crash. Did you report this on the NTR github as an issue?

Derek1199, 2017-06-28:
It's the plugin itself probably though. Also sometimes , this is extremely
rare, but svc_controlmemory when running bootNTR also fails. After running
it a popup in home menu will say svc_controlmemory failed. This has only
happened twice though. So I should put it as a issue On the plugin page as
well.

Margen67, 2017-06-28:
@DeathAngel74 The issues tab was removed from the repo

T3CHNOLOG1C, 2017-06-28:
@Derek1199 try reverting to stable build and see if that fixes it, if not, it's probably a plugin / ntr problem

Hikari-chin, 2017-06-28:
This is an issue with NTR, not Luma.

closed 2017-06-28 by Hikari-chin

Derek1199, 2017-06-28:
OK.

Derek1199, 2017-06-28:
Apparently it was luma conflicting with NTR or the plugin. It's fixed now, no longer getting the error. Not sure what caused it oh well. Thanks