LumaTeam/Luma3DS #669: WHEN hair house homebrewn Game of the Error black farms freacy deluxe fabric

ysneshy, 2017-06-28:
WHEN hair house homebrewn Game of the Error black farms freacy deluxe fabric 
 my FW 11.4.0-37U  luma 8.0 bootloader 9

Help whenever I try to read qr code also of this error, I had used armo9 to unlock in the past and it went wrong, I restored fw and formatted ds and installed bootloader, now it is on 11.4.0.37 with this error that does not let me use homebrewn
An exception occurred

Processor: ARM11 (core 1)
Exception type: prefetch abort (svcBreak)
Current process: loader

R0 00000001 R1 FFF9180C
R2 FFF90F70 R3 FFF9CF1C
R4 D9004587 R5 0FFFFF78
R6 140058C8 R7 00000001
R8 0FFFFEE8 R9 14005470
R10 00000D54 R11 00000001
R12 00000000 SP 0FFFFE20
LR 18001BC8 PC 140034E8
CPSR A0000010 FPEXC 00000000
You can find a dump in the following file:
dump/arm11/crash_dump00000020.dmp

Press an button to shutdown

genehung2003, 2017-06-28:
I fixed the same exception problem for heroes of ruin in no time.
Update your luma through luma updater to the latest hourly build, not stable build
Plz report if it works or not

Hikari-chin, 2017-06-28:
Can't reproduce this, homebrew applications/games/etc. seem to work fine for me. I would test the exact game in question, but the English used is almost unreadable; could you specify exactly what application/game/whatever you're having an issue with?

vaguerant, 2017-06-29:
Issue is talking about using Freakyforms Deluxe (the commercial game) to launch homebrew via QR code, i.e. [freakyhax](https://plutooo.github.io/freakyhax/) by plutoo. When they try to scan the QR code, Luma3DS crash dumps.

ysneshy, 2017-06-29:
I deleted everything and reinstalled nand to 9.2 and to reinstall everything from scratch but I do not know why still the error arm9 when I try to open homebrewm by the sound rax is also giving error foda, I want to take arm9 out of my life that sucks

fincs, 2017-06-29:
hax payloads aren't supported anymore due to changes in the memory layout that had to be done to accomodate Rosalina. It doesn't matter since Rosalina has a built-in feature to load homebrew in the 3dsx format, therefore removing the need for any exploit game such as Freakyforms.

closed 2017-06-29 by fincs

ysneshy, 2017-06-29:
I wanted to ask for help from you, today, manhan 3ds does not turn on, the blue led button stays on for 1 second and turns off and I tried to access start + power select + power and nothing appears, jaz refiz sd card several times and nothing happen

Hikari-chin, 2017-06-29:
You moved or deleted boot.firm from you SD card, or have an invalid one. Download Luma 8.1 and put its boot.firm on the root of your SD card. But sorry guys, the issue tracker isn't for tech support. Take it to the NH Discord.