LumaTeam/Luma3DS #671: Picking 'Power off' sometimes reboots a system instead

SvenDaHacker64, 2017-06-28:
**System model:**
New Nintendo 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.4.0-37E

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap v1.2

**Luma3DS version:**
e67c6ed

**Luma3DS configuration/options:**

Default EmuNAND: 1 (not really using it lol)

Screen brightness: 3

Splash: Before

PIN lock: Off

New 3DS CPU: Clock+L2

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
I was able to Power off my 3DS successfully using "Rosalina" Menu, but it commonly restarts my device, acting like it reboots itself.

**Steps to reproduce:**

1. Opened "Rosalina" Menu (L+↓+Select , preferably R+Start)

2. Went to Miscellaneous options... → Power off → tapped A

**Dump file:**

None

Hikari-chin, 2017-06-28:
Can't seem to reproduce this, powered off ~10 times via Rosalina and didn't get a reboot once. Are you sure you're not just accidentally hitting Reboot instead of Power off? If not, try the latest commit.

closed 2017-06-28 by Hikari-chin

TuxSH, 2017-06-28:
Same as @Hikari-chin

SvenDaHacker64, 2017-06-28:
@Hikari-chin 
Oh, nevermind. I think it was caused by NTR CFW, not Luma3DS itself. Sorry.

Here's a video of attempting to power off after running BootNTR Selector: https://www.youtube.com/watch?v=xfcAzPwLXks