LumaTeam/Luma3DS #674: (Enhancement) Virtaul RAM for the O3DS/2DS or maybe N3DS

Bolt23, 2017-06-28:
As the amount of RAM present in O3DS/2DS is only 64 MB maybe we can have and option like the L2+Clock feature in N3DS that they may have VRAM present on the internal memory of the 3DS since we have access to Custom NCSD through godmode9 and @ApacheThunder's System memory increased to 30000 blocks is the proof! VRAM will also cause the games (especially EMM games) will work flawlessly! Plus HorizonMOD may also have increased frame rates.

Hikari-chin, 2017-06-28:
Closed, reporter misunderstands what flash memory and RAM actually are.

closed 2017-06-28 by Hikari-chin

ahoischen, 2017-06-29:
@Bolt23 I think you should continue discussion of this feature here rather than in the Rosalina compilation. And please read more on the subject.