LumaTeam/Luma3DS #675: Bug with HBL Payload

maorninja, 2017-06-28:
**System model: Old3DS**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable): 11.4-37U Sysnand**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS): (shouldn't this be redundant?) B(S**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version: 8.0**

[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (before)

PIN lock: (none)

New 3DS CPU: (don't own)

--

Autoboot EmuNAND: (Nope)

Use EmuNAND FIRM if booting with R: (No)

Enable loading external FIRMs and modules: (Yes)

Use custom path: (This is an old thing, but with this thingy, it shouldn't)

Enable game patching: (Yes)

Show NAND or user string in System Settings: (Yes)

Show GBA boot screen in patched AGB_FIRM: (Yes)

Patch ARM9 access: (No)

Set developer UNITINFO: (No)

--

**Explanation of the issue:**

Once booting from CTRNand, and still have your SD card inserted, it will look for hax payloads. This also means that what you said about hax payloads not working is partially incorrect, as Puffle Paddle 3DS and Freeshop works, but FTPD doesn't.

**Steps to reproduce:**

1. Boot up 3DS with out SD.

2. Immediately put back the SD card inside.

3. Boot up HBL.

**Dump file:**

maorninja, 2017-06-28:
Also, in your wiki, you said that there are no boot launchers for B9S.

https://gbatemp.net/threads/release-the-bootctr9-release-thread.435347/

Hikari-chin, 2017-06-28:
Compatibility issues like you mentioned with FTPd are the exact reason why we mentioned that we don't support *hax payloads. That they work at all doesn't mean that should be used or that they're officially supported.

closed 2017-06-28 by Hikari-chin

maorninja, 2017-06-28:
No, as in, it actually using hax payloads is the problem.

Hikari-chin, 2017-06-28:
You can get the same behavior by changing your target title to something other than hblauncher_loader and then opening it. I wasn't able to reproduce this behavior unless I did that. Since the boot.firm on CTRNAND and SD are separate payloads, you've got to configure your target title twice, once when booting from SD and once when booting from CTRNAND. That is probably the source of your issue here.

maorninja, 2017-06-28:
Well, on 8.0, it doesn't save the actual settings. and I haven't updated because I only found out now about it.

So its within the CTRNand that is the problem. I didn't change the default target title.

Hikari-chin, 2017-06-28:
Try updating, then, because I tried to reproduce it and was not successful.

DeathAngel74, 2017-06-28:
Try updating to luma 8.1 and hb menu 2.0+.3dsx files.

maorninja, 2017-06-28:
I already have 2.0 and 1.0 hbl menu. I use FBI to switch between the two, when I use other CFW (ReiNAND B9S mode).

And I updated to 8.1 and this issue is fixed. You can keep it closed.

Hikari-chin, 2017-06-28:
Thanks for the report, @maorninja.

fincs, 2017-06-28:
hbmenu 2.0 is completely compatible with older 3dsx launching methods such as hax 2.x. There should be no reason to use older versions of hbmenu.

maorninja, 2017-06-28:
I use the older versions because I like the grid design from Gridlauncher.

Maybe you might think I'm retarded for saying this, but I also like the water.

Margen67, 2017-06-29:
@maorninja Maybe Gridlauncher could be updated to 2.0?