LumaTeam/Luma3DS #680: mcu errors if LED state changes >16 times while Rosalina is open

vaguerant, 2017-06-29:
**System model:**

New 3DS (XL)

**SysNAND version (+emu/redNAND version if applicable):**

11.1.0-34U SysNAND, 9.2-something EmuNAND (no longer actively in use)

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: ( 1 )

Screen brightness: ( 2 )

Splash: ( Off )

PIN lock: ( Off )

New 3DS CPU: ( Clock+L2 )

--

Autoboot EmuNAND: ( Off )

Use EmuNAND FIRM if booting with R: ( Off )

Enable loading external FIRMs and modules: ( Off )

Use custom path: ( **I remain unsure that this feature still even exists in current builds** )

Enable game patching: ( On )

Show NAND or user string in System Settings: ( On )

Show GBA boot screen in patched AGB_FIRM: ( On )

Patch ARM9 access: ( Off )

Set developer UNITINFO: ( Off )

--

**Explanation of the issue:**

Changing the LED state 17 times while Rosalina is open causes mcu to throw an error. I can't confirm this works for all LED changes, but the most convenient one to check is the charging LED.

**Steps to reproduce:**

1. Plug/unplug power repeatedly, changing LED state 16 times.

2. Plug/unplug for the 17th time.

**Dump file:**
No dump file, I assume because it's an error, not a crash. Presumably some kind of overflow?

```
An error occurred (ErrDisp)

Error type:       generic

Process ID:       11
Process name:     mcu
Process title ID: 0x0004013000001f02

Address:          0x0010366c
Error code:       0xd8606408
```

architdate, 2017-06-29:
Does this happen every time regardless of the time interval between the LED state change?

vaguerant, 2017-06-29:
It's happened with any interval that I've tried, from "over a few seconds" to "over about 20 minutes". I haven't tried any more extreme periods.

piratesephiroth, 2017-06-29:
The wifi led doesn't seem to cause the error

TuxSH, 2017-06-29:
This happens because the `mcu` sysmodule doesn't call `srvReceiveNotification` when there are new notifications and the max. number of pending notifications is reached.

HugaShire, 2017-06-29:
Also, the Custom Path for Luma exist because people use chainloaders to boot luma. That then breaks DS(i)/GBA/Extended Memory Games unless the custom path is on.

TuxSH, 2017-06-30:
@HugaShire not a problem anymore since b9s, you can test booting Luma w/ Luma.

Moreover your comment is offtopic.

vaguerant, 2017-06-30:
@TuxSH In fairness, I started it. My original issue includes a non-sequitur when listing my config:

> Use custom path: ( **I remain unsure that this feature still even exists in current builds** )

I was unsure how to fill it when filing the issue as I don't see that option on the Luma3DS config screen. I probably shouldn't have polluted this issue with that completely unrelated confusion, though.

TuxSH, 2017-06-30:
Oops.

Well time to update the issue template again.

Thanks.

NinjaBoyLao, 2017-07-17:
Confirmed in 8.1 stable, 11.5.0-38U SysNAND.

astronautlevel2, 2017-07-19:
Yes, this is a known bug, and doesn't need confirmation.

It'll be fixed when it's fixed.

Pirater12, 2017-11-24:
@TuxSH Can't we use k11_extension to fix this? [maybe like this?](https://hastebin.com/huruqoxibe.php) 
Here 0x400 and 0x401 are notifications sent using `srvPublishToSuscriber()` by rosalina on menuEnter() and menuLeave().

TuxSH, 2020-04-15:
Closing, tracker cleanup

closed 2020-04-15 by TuxSH