LumaTeam/Luma3DS #681: My EmuNAND still doesn't work...

Clement95, 2017-06-29:
**System model:**

[New 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[11.4.0-37E SysNAND, 11.4.0-37E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap/Sighax]

**Luma3DS version:**

[Luma 3DS 8.1 Stable]

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Before)

PIN lock: (Off)

New 3DS CPU: (Clock+L2)

--

Autoboot EmuNAND: (Off)

Use EmuNAND FIRM if booting with R: (Off)

Enable loading external FIRMs and modules: (Off)

Use custom path: (Off)

Enable game patching: (Off)

Show NAND or user string in System Settings: (On)

Show GBA boot screen in patched AGB_FIRM: (On)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

When I try to boot on EmuNAND I have a black screen, but when I press the POWER button, the POWER Menu appears, when I try to go to HOME Menu by pressing HOME, my 3DS shows "HOME Menu" on bottom screen but it's never showed. (Sorry for bad english)

**Dump file:**

MelonGx, 2017-06-29:
"my 3DS shows "HOME Menu" on bottom screen but it's never showed"
So why you put "show" and "not show" together?
Did it actually show or not?

I consider that your EmuNAND has been already corrupted.
Rebuild it if no savedata should be backed-up.

Clement95, 2017-06-29:
I wanted to say "my 3DS shows the "HOME Menu" text when I press the HOME button", the HOME Menu is never showed but the text "HOME Menu" before, yes.
I want to keep my EmuNAND without formatting it, I'll try with GodMode9

Hikari-chin, 2017-06-29:
Can't reproduce; did you install a modified system title on your emuNAND? If not it's likely just been corrupted, check your SD card for damage.

closed 2017-06-29 by Hikari-chin

Clement95, 2017-06-29:
I think that's why I put system titles of my SysNAND to unbrick my
EmuNAND, thank you for your help !

Clement95, 2017-06-29:
My problem is solved ! I unbricked my EmuNAND by copying the "dbs" and "titles" folders of my SysNAND to my EmuNAND