LumaTeam/Luma3DS #682: "An Exception Occurred" error

Askelands, 2017-06-29:
**System model:**

[New 3DS XL]

**SysNAND version (+emu/redNAND version if applicable):**

[11.4.0-37U SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap 1.2]

**Luma3DS version: 8.1 stable**

**Luma3DS configuration/options:**

Default EmuNAND: ( 1 )

Screen brightness: (4 )

Splash: ( off)

PIN lock: ( off)

New 3DS CPU: (off )

Autoboot EmuNAND: (off )

Use EmuNAND FIRM if booting with R: (off )

Enable loading external FIRMs and modules: ( off)

Use custom path: (off )

Enable game patching: (off )

Show NAND or user string in System Settings: ( ON )

Show GBA boot screen in patched AGB_FIRM: (off )

Patch ARM9 access: (off )

Set developer UNITINFO: (off )

Disable ARM11 exception handlers (off)

--

**Explanation of the issue:**

Getting "An Exception Occurred" error screen after updating to latest Luma stable version 

**Steps to reproduce:**
Happened right after updating Luma and powering off, and then a couple more times intermittently when powering off. 

**Dump file:**
Attached
[dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1111922/dumps.zip)

Gliscor, 2017-06-30:
I got an error like this 2 times (dumps attached):

1: When exiting 2DS Configuration menu after configuring NDS internet settings.
2: While was playing Pokémon Moon Online in the Festival Plaza. 

System model:
[Nintendo 2DS]

SysNAND version (+emu/redNAND version if applicable):
[11.4.0-37U SysNAND]

Entrypoint (How/what you're using to boot Luma3DS):
[Boot9Strap 1.2]
Luma3DS version: 8.0 stable

Luma3DS configuration/options:
Default EmuNAND: ( 1 )
Screen brightness: (4 )
Splash: ( off)
PIN lock: ( off)
New 3DS CPU: (off )
Autoboot EmuNAND: (off )
Use EmuNAND FIRM if booting with R: (off )
Enable loading external FIRMs and modules: ( off)
Use custom path: (off )
Enable game patching: (off )
Show NAND or user string in System Settings: ( ON )
Show GBA boot screen in patched AGB_FIRM: (off )
Patch ARM9 access: (off )
Set developer UNITINFO: (off )
Disable ARM11 exception handlers (off)

[dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1113833/dumps.zip)

VKLink, 2017-06-30:
**System model:**

[OLD 3DS XL]

**SysNAND version (+emu/redNAND version if applicable):**

[11.4.0-37U SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap 1.2]

**Luma3DS version:** 

8.1 stable release

**Luma3DS configuration/options:**

Default EmuNAND: ( 1 )

Screen brightness: (4 )

Splash: ( off)

PIN lock: ( off)

Autoboot EmuNAND: (off )

Use EmuNAND FIRM if booting with R: (off )

Enable loading external FIRMs and modules: ( off)

Enable game patching: (**ON**)

Show NAND or user string in System Settings: (**ON**)

Show GBA boot screen in patched AGB_FIRM: (off )

Patch ARM9 access: (off )

Set developer UNITINFO: (off )

Disable ARM11 exception handlers (off)

--

**Explanation of the issue:**

_Same Issue as Askelands:_

Getting "An Exception Occurred" error screen after updating to latest Luma stable version.
Never happened before, when i was on 8.0.1

Steps to reproduce:
Happened right after updating Luma and powering off, and then a couple more times intermittently when powering off.

Dump file:
i'll upload it later, promise!

Juelz89, 2017-07-01:
System model:

[OLD 3DS XL]

SysNAND version (+emu/redNAND version if applicable):

[11.4.0-37U SysNAND]

Entrypoint (How/what you're using to boot Luma3DS):

[Boot9Strap 1.2]

Luma3DS version:

8.1 stable release

Luma3DS configuration/options:

Default EmuNAND: ( 1 )

Screen brightness: (4 )

Splash: ( off)

PIN lock: ( off)

Autoboot EmuNAND: (off )

Use EmuNAND FIRM if booting with R: (off )

Enable loading external FIRMs and modules: ( off)

Enable game patching: (off)

Show NAND or user string in System Settings: (ON)

Show GBA boot screen in patched AGB_FIRM: (off )

Patch ARM9 access: (off )

Set developer UNITINFO: (off )

Disable ARM11 exception handlers (off)

Explanation of the issue:

Getting "An Exception Occurred" error screen after starting the fbi.3dsx file in homebrew launcher menu

I was using the guide from https://3ds.guide Section III and IV
[https://3ds.guide/finalizing-setup](url)

Dump file:
[dump.zip](https://github.com/AuroraWright/Luma3DS/files/1117381/dump.zip)

Hikari-chin, 2017-07-02:
Can't reproduce any of the crashes mentioned here.

closed 2017-07-02 by Hikari-chin

Askelands, 2017-07-07:
[image]

Just happened again. Just updated to latest stable 8.1-31e2243 and as soon as I hit start to reboot, it crashed.