LumaTeam/Luma3DS #684: Rosalina Input Issue

ZenIsBestWolf, 2017-06-29:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]
Old 3DS XL (United States)
**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]
11.4.0-37U SysNAND
**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]
Boot9Strap 1.2
**Luma3DS version:**

[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

8.1
**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: (Before)

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
Somtimes, the Rosalina menu will not take input in if input is given at a relatively fast speed.

*Sometimes*! Not always. Please be patient while reproducing.

**Steps to reproduce:**

1. Open Rosalina

2. Input selections very fast, like spamming the Down button.

**Dump file:**

None.

DMSalesman, 2017-06-30:
Unable to reproduce on European Old 3DS.

T3CHNOLOG1C, 2017-07-05:
Unable to reproduce on US Old 3DS, Luma v8.1-89fca388

Jan200101, 2017-07-05:
Unable to reproduce on Old 3DSEu

Hikari-chin, 2017-07-05:
I can't reproduce this, no matter how long I try; I'm on a US New 3DS XL. It's likely your buttons are worn or damaged.

closed 2017-07-05 by Hikari-chin