LumaTeam/Luma3DS #688: Error with Apps set as homebrew launcher

Owta, 2017-06-30:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.1

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (3)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (Clock + L2)

--

Autoboot EmuNAND: (Off)

Use EmuNAND FIRM if booting with R: (Off)

Enable loading external FIRMs and modules: (On)

Use custom path: (Off)

Enable game patching: (On)

Show NAND or user string in System Settings: (On)

Show GBA boot screen in patched AGB_FIRM: (Off)

Patch ARM9 access: (Off)

Set developer UNITINFO: (Off)

--

**Explanation of the issue:**

When booting Homebrew launcher via manually set app, and closing homebrew launcher via HOME Menu, the 3DS crashes.

Video of the error: https://youtu.be/Z2y-4edd2gI

**Steps to reproduce:**

1. Set an app to open homebrew launcher

2. Open then close the app via the HOME Menu

**Dump file:**
[crash_dump_00000002.zip](https://github.com/AuroraWright/Luma3DS/files/1116490/crash_dump_00000002.zip)

Margen67, 2017-06-30:
https://github.com/fincs/new-hbmenu/releases

Owta, 2017-06-30:
should've clarified, im already using the new homebrew launcher

closed 2017-06-30 by TuxSH

Hikari-chin, 2017-07-01:
Can't reproduce.