LumaTeam/Luma3DS #689: Things Drawn on the Bottom softlock Rosalina

Jan200101, 2017-06-30:
**System model:**

Old 3ds XL

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

https://github.com/AuroraWright/Luma3DS/commit/31e2243c5cb7736ff30be3781dfd2c854c2f61aa

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: (After)

PIN lock: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: (X)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

If another app is drawing a object onto the bottom screen (like NTR or similar, this does not limit itself to NTR) Rosalina softlocks the console on access

**Steps to reproduce:**

1. Start anything that draws on the bottom screen ([Ntr Overlay Example](https://github.com/44670/ntr_overlay_samples/releases) for example) 

2. Open Rosalina

Sonlen1414, 2017-06-30:
Can confirm, all settings match above aside from Clock+L2 on a New 3DS XL, opening Rosalina while using a NTR Plugin that draws to the bottom screen locks the conole and forces a shutoff holding the power button.

SvenDaHacker64, 2017-07-01:
That happens to me everytime, because Rosalina and NTR CFW collide each other and causes to freeze the game completely.

fox8091, 2017-07-03:
Is it only with NTR? If so, that's likely because two different apps are trying to stall the system, resulting in the entire console softlocking.

Jan200101, 2017-07-03:
Its been confirmed to happen with NTR but we cant say for sure that normal applications arent affected because of the lack of Applications that draw a permanent object on screen

TuxSH, 2017-07-03:
With the lack of other examples than NTR I'll assume it's a NTR issue.

Hikari-chin, 2017-07-05:
I think this is an issue with certain plugins you're using. I can't reproduce this with NTR CFW 3.5, I can trigger both the Rosalina and NTR menu without locking the system, likewise with the Gateshark2NTR menu in certain applications.

Hikari-chin, 2017-07-05:
Ro

closed 2017-07-05 by Hikari-chin

Jan200101, 2017-07-05:
@Hikari-chin Not the NTR menu.
If something draws on the normal screen like the example plugin mentioned in OP it freezes

Hikari-chin, 2017-07-05:
Yes, I'm saying that, NTR and Gateshark2NTR both draw on the screen, and don't freeze, so it's probably an issue with that specific plugin.