LumaTeam/Luma3DS #693: Arm11 Crash

Tkompuras, 2017-07-02:
**System model:**
Old 3ds

**SysNAND version (+emu/redNAND version if applicable):**
11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
8.1 Stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (-)

--

Autoboot EmuNAND: (-)

Use EmuNAND FIRM if booting with R: (-)

Enable loading external FIRMs and modules: (-)

Enable game patching: (-)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (-)

Patch ARM9 access: (-)

Set developer UNITINFO: (-)

Disable ARM11 exception handlers: (-)

--

**Explanation of the issue:**
Recently, I changed from A9LH to B9S so I could update Luma to the latest stable version. I followed the guide step-by-step and I think it worked. Then I wanted to play a game, namely "Ever Oasis", so I launch the game and I get the screen you can see below as well as a arm11 crash dump which I have included.

I also have tried enabling the option "Disable ARM11 exception handlers".

**Steps to reproduce:**

1. Power up the console

2. Launch the Game

**Dump file:**
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1117971/crash_dump_00000000.zip)
[image]

Hikari-chin, 2017-07-02:
Duplicate of several other issues. Ever Oasis requires locale emulation if you're using the EUR versio. on a US console or vice versa.

closed 2017-07-02 by Hikari-chin

Tkompuras, 2017-07-02:
Thanks, sorry I didn't notice another thread like this.