LumaTeam/Luma3DS #694: Rosalina menu doesn't work with DSi/GBA titles

henke37, 2017-07-02:
**System model:**
Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**
11.0.0-33E

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9strap

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: ( 1)

Screen brightness: ( 4)

Splash: ( off)

PIN lock: (off )

--

Autoboot EmuNAND: ( x)

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

Trying to open the Rosalina menu does not work if the system is running a DSi or GBA title.

**Steps to reproduce:**
0. Verify that the Rosalina menu can be opened in normal titles
1. Launch a DSi ware or GBA title
2. Try and open the Rosalina menu

henke37, 2017-07-02:
Updated to 8.1, made no difference.

TuxSH, 2017-07-02:
Of course not, DSi/GBA titles run on baremetal.

closed 2017-07-02 by TuxSH

HugaShire, 2017-07-03:
Of course, the games run it ARM 7 mode (GBA and DS(i)). Most of the OS isn't running.