LumaTeam/Luma3DS #695: Super Mario Maker NTR Plugin Crash When Using Certain Cheats

JGamez065, 2017-07-02:
System model:
Old 3ds XL
SysNAND version (+emu/redNAND version if applicable):
11.4.0-37U SysNAND
Entrypoint (How/what you're using to boot Luma3DS):
Boot9Strap

Luma3DS version:
31e2243

Luma3DS configuration/options:
Default EmuNAND: (1)
Screen brightness: (4)
Splash: (OFF)
PIN lock: (OFF )
--
Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Enable game patching: ( )
Show NAND or user string in System Settings: ( )
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )
Disable ARM11 exception handlers: ( )
--
Explanation of the issue:
System model:
Old 3ds XL
SysNAND version (+emu/redNAND version if applicable):
11.4.0-37E SysNAND
Entrypoint (How/what you're using to boot Luma3DS):
Boot9Strap
Luma3DS version:
31e2243
Luma3DS configuration/options:
Default EmuNAND: ( )
Screen brightness: ( )
Splash: (After)
PIN lock: ( )
--
Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Enable game patching: (X)
Show NAND or user string in System Settings: (X)
Show GBA boot screen in patched AGB_FIRM: (X)
Patch ARM9 access: ( )
Set developer UNITINFO: ( )
Disable ARM11 exception handlers: ( )
--
Explanation of the issue:
https://gbatemp.net/threads/release-super-mario-maker-ntr-plugin.451710/ I frequently used this plugin before Boot9strap. Today when I tried it, Luma would give me a dump error if attempting to use "Coordinates Modifier" before updating to boo9strap it worked perfectly. (Using the unofficial version and was directed by the people from the discord server to report this error. 

Steps to reproduce:
1. Open Super Mario Maker
2. Press "Select" to open cheat menu and press a to turn cheats "ON"
3. Find "Coordinates Modifier" cheat in menu and make it select "ON"
4. Back out and press y and direction to attempt to move which crashes. (Will not crash if other cheats are being used and not using coordinates modifier even if enabled.)

[dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1118165/dumps.zip)

maorninja, 2017-07-03:
That's a problem with the plugin itself/NTR and not Luma/Rosalina.

Please close this.

closed 2017-07-03 by Hikari-chin

Fitsneezy, 2026-04-19:
nevermind i fixed it i had to delete all the cheats