LumaTeam/Luma3DS #696: TWL cart seemed break again on N3DS EmuNAND

MelonGx, 2017-07-03:
Tested on both 8.1 and 7.1 a6dcce0.
TWL cart worked on both SysNAND but crashed on both EmuNAND.
Not tested with AGB & DSiWare.

Settings:
EmuNAND 1
Brightness 4
Splash After
Pin lock Off
N3DS CPU Clock+L2
Autoboot EmuNAND
Enable exception handlers (it auto-enabled on 8.1)

Hikari-chin, 2017-07-03:
Have you ever installed a parched TWL_FIRM? Try the ["DSi / DS functionality is broken after completing the guide" section of the guide](https://3ds.guide/troubleshooting).

MelonGx, 2017-07-03:
@Hikari-chin Never. Should I proceed this?

Hikari-chin, 2017-07-03:
Yes, please do so. It can't hurt.

MelonGx, 2017-07-03:
Patching TWL CIAs crashed EmuNAND.
Now it can't be booted anymore.

Hikari-chin, 2017-07-03:
Uh, I think you misunderstood what I said. I said "Try the ["DSi / DS functionality is broken after completing the guide" section of the guide](https://3ds.guide/troubleshooting)", which installs the original TWL CIAs. 

Did you go find patched ones elsewhere? If so, yeah, Luma doesn't support patched system titles; causes issues with its built-in patches. You need to install the clean CIAs in the guide I linked above and nothing else.

MelonGx, 2017-07-04:
I never have any patched things.
But re-installing original CIA broke the EmuNAND.

Hikari-chin, 2017-07-04:
User error in that case, try the NH Discord.

MelonGx, 2017-07-04:
No need for that. Rebuilding the EmuNAND.

closed 2017-07-04 by MelonGx

MelonGx, 2017-07-04:
@Hikari-chin I have my way to recover things and find out the fact of this issue. But I dislike your last attitude after crashed an EmuNAND from your suggestion.

Hikari-chin, 2017-07-04:
So, I'll clarify: The issues aren't for technical support for problems deriving from user error.

Hikari-chin, 2017-07-04:
Confirmed, this was user error:

```
MelonGx - Today at 8:40 PM
Installing original TWL CIAs with Rosalina enabled Luma breaks TWLN partition with an ErrDisp occurred during the 4th CIA installation
BTW, injecting SysNAND TWLN fixes the issue
junko - Today at 8:40 PM
@Sonlen Thank you. How weird.
Sonlen - Today at 8:40 PM
the weird part is start reboots it, not sure why thats a thing o.O
junko - Today at 8:40 PM
And @MelonGx, how did you install those CIAs, exactly?
MelonGx - Today at 8:41 PM
BigBlueMenu
not FBI
junko - Today at 8:41 PM
Yeah...use FBI
MelonGx - Today at 8:41 PM
not DevMenu
junko - Today at 8:41 PM
BBB is an old rebadged DevMenu
You should've mentioned you encountered a crash while installing CIAs in the original issue
That's what "broke" your emuNAND
Not installing the CIAs themselves
And it wasn't my fault, either
MelonGx - Today at 8:43 PM
OK I have bad communication to everyone
Sonlen - Today at 8:47 PM
@junko and just for the record, DSiWare has the same reaction, freeze on black screen upon launch, and start reboots the console(edited)
installed to both Sys and Emu
junko - Today at 8:47 PM
@Sonlen, never installed patched TWL_FIRM CIAs, right?(edited)
Sonlen - Today at 8:47 PM
I don't think I did
MelonGx - Today at 8:47 PM
My case may be a System Transfer issue
Sonlen - Today at 8:47 PM
because I never tried on Emu
lemme do that and let you know what happens?
junko - Today at 8:48 PM
Sure.
Sonlen - Today at 8:48 PM
unless you know it will fix it without a doubt lol
junko - Today at 8:48 PM
If the issue persists, could you open an issue about it on Luma's repository, @Sonlen?
Sonlen - Today at 8:48 PM
will do
astronyaa 🌈 - Today at 8:49 PM
What's going on?
MelonGx - Today at 8:50 PM
Nothing but my fault
Sonlen - Today at 8:50 PM
just testing an issue
MelonGx - Today at 8:51 PM
Anyway I'll avoid creating issue threads for not confusing people
Sonlen - Today at 8:52 PM
@junko yeah, TWL_FIRM CIAs fixed it right up, I can't believe I didn't think of that helper role, right
junko - Today at 8:52 PM
@Sonlen Thank you.
Sonlen - Today at 8:53 PM
no problem!
```