LumaTeam/Luma3DS #697: GDB Stub memory breakpoints not working. (IDA Pro)

mariohackandglitch, 2017-07-04:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (off)

PIN lock: (off)

New 3DS CPU: (off)

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

GDB Stub memory watchpoints not working with IDA pro, I can see the desired memory address change with the hex view, but it doesn't break when it changes.

**Steps to reproduce:**
1. Open IDA database.
2. Enable Luma debugger
3. Attach IDA to the current running process.
4. Jump to the desired address.
5. Debugger -> Breakpoints -> Add breakpoint (This will default the breakpoint to Execute)
6. Debugger -> Breakpoints -> Breakpoint list
7. Edit the breakpoint to be Read/Write instead of execute, as it defaults to Execute. (Break option is enabled)
8. The game doesn't break when the memory address is changed/read.

**Dump file:**
N/A

TuxSH, 2017-07-04:
It works for me, although this lacks lack documentation:

* Make sure it's a hardware RW breakpoint (maximum 2, this is a hardware limitation)
* Make sure the byte size of the breakpoint is exactly the correct one (still a hardware limitation)

mariohackandglitch, 2017-07-04:
Here you can see that I'm using the settings you mentioned:
http://i.imgur.com/ZpQ54py.png
( the screenshot shows Write is disabled but I enabled it afterwards).
I'm trying to debug Mario Kart 7, if you want more info, ask me about it. I can also record the whole process if you want.

RyDog199, 2017-07-04:
I can't seem to get memory watchpoints or breakpoints to work either. 
I'm using the same IDA settings as @mariohackandglitch but this time, I'm trying to use it on an Animal Crossing New Leaf copy.

mariohackandglitch, 2017-07-04:
Actually, @TuxSH you said to make sure it's a hardware breakpoint, but those are the ones which aren't working. Making a non hardware breakpoint (code breakpoint) works fine, however, making a execute hardware breakpoint doesn't work.

Slattz, 2017-07-04:
Hardware breakpoints also don't work for me.
As @mariohackandglitch said above, a non hardware breakpoint (code breakpoint) works but a hardware breakpoint doesn't work. 
I'm using Luma 8.1 on a 11.4 EUR o3ds. I'm trying to use it on Animal Crossing New Leaf (EUR Cart), like @RyDog199.

TuxSH, 2017-07-05:
> making a execute hardware breakpoint doesn't work.

Those are not implemented (shitty hardware limitations), execute breakpoints are software-breakpoints only.

I meant watchpoints.

mariohackandglitch, 2017-07-06:
That wouldn't be a problem if R/W breakpoints were working.

TuxSH, 2017-07-08:
They are.

mariohackandglitch, 2017-07-08:
Could you send your IDA database and version to see if that's the problem? (The debugger settings will be saved in the database)
Do I really have to make a video to convince you they aren't working? I'm not the only person who is having this problem, so there has to be something that we aren't doing the same way so they only work in your side.

Slattz, 2017-07-08:
Yea, hardware R/W breakpoints just don't work for me either, I've tried loads of things but they never work.
It would be great if we could get some details on your setup or something, like said above :)

Nanquitas, 2017-07-09:
Watchpoints are not working for me either.
Breakpoint on code are working fine though.

Example with Zelda A Link Between World (EUR):

Current heart address: 0x168B7444 (u8)
Max heart address: 0x168B7445 (u8)
The function that handle the heart (do damage, potion etc) starts at 0x00149BD4.

Creating a breakpoint at 0x00149BD4 will work and [here's a screen.](http://i.imgur.com/1tfIpJ3.png)
As you can see, R0 is the address of the heart(0x168B7444), and 0x00149BDC instruction will load a byte from this address.
So theoretically, when the instruction is executed, a watchpoint made at 0x168B7444 shoud be triggered right ?

But it's not. The watchpoints are never triggered.
[Watchpoint made](http://i.imgur.com/ZWJRHis.png)
[Watchpoint settings](http://i.imgur.com/piuS5qr.png)

I'm available for more testing / infos if you need (here, GBATemp, Discord, your choice).
Thanks.

movr1r1, 2017-07-09:
@Nanquitas are sure ?
"Watchpoints" work for me, I have made some patch on some games with that, so I don't see why it does not work for you.

Nanquitas, 2017-07-09:
Yeah, I can't make it work with the few games I tried.
Code breakpoint works fine but not watchpoint.

Could it be that I overlooked something ?

@movr1r1 : Do you mind sharing your settings ?

movr1r1, 2017-07-09:
Sure, when i'll be back from work.

Nanquitas, 2017-07-09:
Awesome, thanks ! :)

mariohackandglitch, 2017-07-10:
@movr1r1

Nanquitas, 2017-07-12:
So, since I wasn't able to get any precise infos on why it doesn't work, I added a logger to Luma3DS to see what happens.

[Here's a screen of the logs.](http://i.imgur.com/OZ0tfeG.jpg)

So, we can see that the watchpoint is correctly set, but it's never triggered.
It should create a new log if a new DBGEVENT_EXCEPTION is detected and there's nothing (I took several hits, so the address is definitely read / written).

What could prevent the watchpoint from being triggered ?

[Software breakpoint on code works fine](http://i.imgur.com/RPutIw7.jpg)

Also, for those that can make it work, could you please tell me which game and address did you tried ?
I'd like to test those too.

mariohackandglitch, 2017-07-13:
(Mentioning @TuxSH as he probably doesn't get notifications of closed threads)
@Nanquitas also tried with GDB Stub 8.0, and he gets the same result.
My guess is that something else is using the breakpoints.

TuxSH, 2017-07-19:
Watchpoints work fine. For example, I've set them on sm, hid and other system modules and _they work fine_.

Nanquitas, 2017-07-19:
I don't use NTR CFW when setting them.

I've set them on games and nothing. I tried with 2 differents 3DS, same results.

The same watchpoints works fine when connected to Citra, but Citra is slow in debug mode.

I also tried with a clean system, from an unmodified 9.2 (Created an emunand with a nand backup) and it didn't worked either.

I tried many games but nothing.
It's too bad because it's a really nice feature that got me excited but well...

Also, I confirmed that the DSCR, WCR, WVR, BCR & BVR are correctly set, but the watchpoints are just not triggered. I can't find why.

I'll try them on a system module if it can help.
Any address (on hid ?) you can give me to test ?

astronautlevel2, 2017-07-20:
It seems like Tux pushed a commit 15 hours ago that changes the behavior of watchpoints: https://github.com/AuroraWright/Luma3DS/commit/3907c46980c3f9deb54520c743cc2530b77637b9

Does this fix the issue?

mariohackandglitch, 2017-07-28:
@astronautlevel2 Still not working (same behavior), tested with Mario Kart 7.

TuxSH, 2017-10-14:
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1384886/boot.zip)

Could you try this build?

RyDog199, 2017-10-14:
Still doesn't seem to work.
I tried setting R/W trace to memory address 0x31F2DBF0 in Animal Crossing New Leaf (USA 1.5), the address of your inventory, but nothing triggered when making sure that the game were to write something to this address. NTR Dex works fine for this address, so I know that it's not the address's fault.

Nanquitas, 2017-10-14:
First of all, I'd like to say thanks for keep trying and working on this, it's really appreciated. 👍

However, I just tried with Zelda OOT and no watchpoints got triggered.

Nanquitas, 2017-11-22:
This issue can be closed as it's now solved thanks to 506b16db378f8a0e3114dfc97d0a313038814c88.

closed 2017-11-22 by Hikari-chin