LumaTeam/Luma3DS #699: Pokemon Art Academy can not be patched.

FlameBeam, 2017-07-04:
System model: new3DS

SysNAND version: 11.4.0-37J

Entrypoint: Boot9Strap

Luma3DS version: 8.1 (game patch enabled)

Explanation of the issue:
I tried to make a game patch for Pokemon Art Academy, but it can't work.
I tested with JPN(00040000000CF900) and USA(00040000000D0A00) version gamecart, but they all can not be patched.

AuroraWright, 2017-07-04:
Are you referring to LayeredFS? What happens?

FlameBeam, 2017-07-04:
Yep, I referring to LayeredFS. 
If I put a romfs folder as a LayeredFS patch, nothing happens in game.
But exefs.bin can be patched. 
I also try the OnionFS, same thing like LayeredFS.

AuroraWright, 2017-07-04:
This should be fixed in the latest nightly, please test

FlameBeam, 2017-07-05:
I tested, the LayeredFS patch can be work for this game now! Thanks!

closed 2017-07-05 by Hikari-chin