LumaTeam/Luma3DS #703: Randomized Pokemon 3DS Roms Don't Work

gui710, 2017-07-05:
**System model:**

o2DS (11.4.0-37E SysNAND)

**Entrypoint:**

Boot9Strap

**Luma3DS version:**

Luma3DS 8.1 Stable (Also Happened with 8.0 Stable)

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

Show NAND or user string in System Settings: (X)

--

**Explanation of the issue:**

I open my extracted, randomized and rebuilt Pokemon .cia ROM, wait a few seconds and it gives me an ARM11 exception handler [prefetch abort (svcBreak)], I've tried with several roms and cartridge dumps from my friends but neither of them works.

**Steps to reproduce:**

1. Decrypt You 3DS Pokemon ROM (Sun In my case)
 
2. Extract it using "HackingToolkit3DS"

3. Randomize Wild Encounters, Trainers and Starters using PK3DS

4. Rebuild the .cia using the same tool as in step 2.

5. Install & Run it

**Dump File:**
[dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1123269/dumps.zip)

sora10pls, 2017-07-05:
Use SaltySD.

gui710, 2017-07-05:
@sora10pls I'd use it but i've tried it before and it's way too confusing...

sora10pls, 2017-07-05:
Every person I've seen rebuild ROMs of Sun/Moon have had issues running them; all you do is replicate the filesystem.

If you don't want to use SaltySD, use Luma's LayeredFS implementation. I do however prefer SaltySD due to much quicker load times.

Example: Sun wild encounter GARC is `/a/0/8/3`.

Copy the `3` file to the following path on your SD Card:
`luma/titles/0004000000164800/romfs/a/0/8/`

You would then just move all the modified GARCs to their proper locations.
`Shop.cro` goes in the `romfs` folder.
ExeFS edits go in `luma/titles/0004000000164800/` as `code.bin`.

Lastly, just make sure you have "Enable game patching" enabled in your Luma config.

Hikari-chin, 2017-07-05:
Can't reproduce. Have played randomized Pokemon ROMs recently, never had an issue.

closed 2017-07-05 by Hikari-chin

gui710, 2017-07-05:
@Hikari-chin Could I be doing anything wrong in the process of randomizing? eitherway i'll just use Luma's LayeredFS...