LumaTeam/Luma3DS #705: FBI>Titles Freezes for both SysNAND and EmuNAND

kamenrider-darkglass, 2017-07-05:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

9.2.0-20U SysNAND (+11.4.0-37U EmuNAND)

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.1 stable 

**Luma3DS configuration/options:**

Default EmuNAND: 1(x)

Screen brightness: (4)

Splash: Off(x)

PIN lock: Off(x)

New 3DS CPU: Off(x)

--

Autoboot EmuNAND: (x)

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
When in either SysNAND or EmuNAND, when I try to access FBI>Titles, the screen freezes. This is a problem because I regularly delete games and their tickets that I don't want to play anymore. 

**Steps to reproduce:**

1. Boot either SysNAND or EmuNAND

2. Go to FBI>Titles

3. Screen Freezes when trying to load all the tiles

4. It doesn't freeze like 1 out of 3 times, so make sure to repeat the steps again when it seems fine at first. 

**Dump file:**

Hikari-chin, 2017-07-05:
Using FBI 2.4.8; can't reproduce. If you're running an FBI nightly, this is likely an issue with it, not Luma.

closed 2017-07-05 by Hikari-chin

AuroraWright, 2017-07-07:
https://github.com/fincs/citro3d/issues/34