LumaTeam/Luma3DS #706: Expanded Mode Terraria causing ARM11 to crash

BYFSartre, 2017-07-06:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U SysNAND

**Entrypoint:**

B9S

**Luma3DS version:**

8.1 stable

**Luma3DS configuration/options:**

Everything (including L2 and CPU) turned off.

**Explanation of the issue:**

Once you open the minimap, the game tends to crash from time to time. There seems to be some time after starting the game that the map works, but after 10 or so minutes, the game crashes upon opening the minimap.

In all sincerity, I'm a noob. Would disabling ARM11 exception handlers fix this? If so, I apologize for opening a ticket, couldn't find info anywhere.

**Steps to reproduce:**

1. Create new singleplayer expanded world in Terraria 3DS (1.0.5 latest.)

2. Load into the world.

3. Wait around 10 minutes, then open minimap (or open it continuously, not sure, it seems to be a random crash)

**Dump file:**

[crash_dump_00000004.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/1127431/crash_dump_00000004.dmp.zip)

Gudf, 2017-07-06:
Did you try with N3DS CPU+L2 activated ? The console may simply not be able to handle expanded worlds with O3DS CPU.

LiquidFenrir, 2017-07-06:
you've got it backwards, it only forces anything when CPU|L2 is on.
And just tried it, no crash in sight. sorry.

closed 2017-07-06 by Hikari-chin

BYFSartre, 2017-07-06:
Thanks for testing it out. I tested with both CPU + L2 on and off, both crashed. I have played for around 2 hours after disabling exception handlers and there have been no crashes.

Could be just me. Also, I don't use NTR, or anything else, with Terraria. I find it odd that it's only happening to me. Thanks for investigating.

Hikari-chin, 2017-07-06:
The exception handlers themselves wouldn't be causing crashes. Also, please read the description on the bottom screen when selecting that option.

BYFSartre, 2017-07-07:
Indeed, I know that disabling the exceptions bars you from posting here, that being said, after doing some more digging, [it seems like it's a Terraria issue](https://forums.terraria.org/index.php?threads/terraria-3ds-1-04-issues-thread.51747/page-38#post-1401920), not a Luma one. 

I'm just leaving this here in case someone runs into the same issue in the future.

Once again, thanks for the support.