LumaTeam/Luma3DS #707: [3DS Game Crash]Heroes of Ruin "RiftWalk" error.

ScytheIronclaw, 2017-07-07:
*I am aware that this issue was brought up with a previously closed post by another user. I wish to confirm this issue.

Hardware: new 3DS XL USA Region
Firmware: 11.2.0-35U

Luma Version:
8.1-557f205
Luma Configurations:
Default Emunand: 1
Screen Brightness: 1
Splash: Off
PIN Lock: Off
New 3DS CPU: Off
Autoboot EmuNAND: False
Use EmuNAND FIRM if booting with R:False
Enable loading external FIRMs and modules:False
Enable game patching:True
Show NAND or user string in System Settings:True
Show GBA boot screen in patched AGB_FIRM:False
Patch AMR9 access:False
Set developer UNITINFO:False
Disable AMR11 exception handlers:False

The issue:
After launching Heroes of Ruin and setting the following initial settings:
Enable Spotpass? NO > Enable User Generated Content? NO > Receive Notifications from Square Enix? NO

After selecting these options the following error is displayed on the bottom screen:

> Error type: logged
> Process ID: 41
> Process name: RiftWalk
> Process title ID 0x0004000000075100
> 
> Address: 0x003ec3dc
> Error Code: 0xc8a0f833

If no one else is getting this error with a retail cart of this game or their own personal dump of this game, it could be possible that the two different CIAs that I downloaded from that ISO site are bad dumps. I'll see if I can track down an official scene release and try building a CIA from a .3DS file

What I tried:
Loading the game with region settings set to SYSTEM DEFAULT > Crash
Loading the game after setting region setting in Luma Locale Changer app to USA EN > Crash

*None of the other games from the last list I have give me an error. Dragon Quest with the OST hack just needs Luma Locale setup, Majora's Mask works fine (so far, haven't played it, just booted it once) USA Dumps of Ever Oasis work fine. SMT Apocalypse doesn't give me DLC loading errors, etc, etc.

Hikari-chin, 2017-07-07:
Sounds like bad dumps, but we don't support piracy tools or websites.

Margen67, 2017-07-07:
>11.2

ScytheIronclaw, 2017-07-08:
I have a few questions about this closed issue.
1 - What would be a valid Issue report? What made this invalid?
2 - I've tried a few different dumps done by a few different users, I can't seem to track down a .3DS dump. If I find and buy the physical cartridge and still encounter this issue, would this be considered valid?

This is the only game so far that has any issues with Luma 8.1
I'm using a 128GB MicroSD Card and aside from a big longer initial home menu load times, I've had no other issues with any games or features.

Also is there some way to check the version of B9S I am using? I wonder if there was a later update to B9S but if I don't know what I got installed I have no idea how to troubleshoot this issue.

Hikari-chin, 2017-07-08:
I actually misread this a bit, my apologies. Luma's exception handlers don't actually occur on the bottom screen in such a way, so this error is likely the title's own exception handlers. Rather than a Luma issue, this is likely laziness by the developers.

Hikari-chin, 2017-07-08:
I confirmed this as an actual issue with Luma. While it's possibly due to laziness on Square Enix London's part, it will be looked into and fixed. Thank you.

commit 7e7ab124 (2017-07-18): Log "logged" errdisp messages in /luma/errdisp.txt, fixes #707
closed 2017-07-18 by TuxSH