LumaTeam/Luma3DS #708: ARM11 core 0 exception occurred while booting to home menu with luma 8.1 on new 3ds

Lopicl, 2017-07-07:
**System model:**
New 3DS

**SysNAND version (+emu/redNAND version if applicable):**
9.0.0-20E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
boot9strap 1.2

**Luma3DS version:**
8.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (Clock+L2)

--

Autoboot EmuNAND: (NO)

Use EmuNAND FIRM if booting with R: (NO)

Enable loading external FIRMs and modules: (NO)

Enable game patching: (NO)

Show NAND or user string in System Settings: (YES)

Show GBA boot screen in patched AGB_FIRM: (YES)

Patch ARM9 access: (NO)

Set developer UNITINFO: (NO)

Disable ARM11 exception handlers: (NO)

--

**Explanation of the issue:**

Everytime I boot the console, i get an ARM11 error on CORE 0 before entering the home menu (data abort)

**Steps to reproduce:**

1. Boot the console with boot9strap 1.2 installed and 8.1 luma3ds boot.firm

**Dump file:**
[crash_dump_00000002.zip](https://github.com/AuroraWright/Luma3DS/files/1130802/crash_dump_00000002.zip)

Hikari-chin, 2017-07-07:
Can't reproduce. [Try deleting your Home Menu extdata](https://3ds.guide/troubleshooting), under "Black screen on sysNAND boot".

closed 2017-07-07 by Hikari-chin

david-zzj, 2017-08-11:
is it solved ？

Lopicl, 2017-08-11:
Yes I solved

david-zzj, 2017-08-11:
can you tell me how is your solution ？

Lopicl, 2017-08-11:
I honestly don't remember, i think i restored my emunand backup on the
sysnand (as i switched from menuhax to boot9strap). If that doesn't work
try deleting your home menu extdata with this guide
https://3ds.guide/troubleshooting

david-zzj, 2017-08-11:
i have try delet ，when boot again ，after creating the homemenu data ，the issue dump screen display again.....