LumaTeam/Luma3DS #710: "enable game patching" option doesn't work!

Berkan2002, 2017-07-08:
I am trying to randomize Pokemon Sun with SaltySD and I did everything right but it is not randomized. I also tried using layeredfs but it doesn't randomize the game aswell. I have boot9strap and have luma 8.1 with the "enable game patching" option checked but it just doesn't work. I really need help so please someone help me with this issue. I have this issue for months now :(

Hikari-chin, 2017-07-08:
Duplicate of other issues. Additionally, SaltySD is unrelated to Luma and we can't provide support for it here.

closed 2017-07-08 by Hikari-chin

Berkan2002, 2017-07-08:
But layeredfs also doesnt work. What do I do?

Hikari-chin, 2017-07-09:
Tested. LayeredFS redirection and code.bin replacement both work for me after using pk3ds. However, the issue tracker isn't the proper place for tech support; try the [NH Discord](https://discord.gg/C29hYvh).