LumaTeam/Luma3DS #711: Rosalina interferes with games requiring more memory like Smash Brothers

SilentHeII, 2017-07-08:
**System model: N3DS XL**

**SysNAND version: 11.4-37U**

**Entrypoint: B9S**

**Luma3DS version: 8.1**

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: (Clock+L2)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
When trying to play Online, loading times are exceptionally longer. Finding matches is also hit and miss sometimes throwing you in an empty room where you wait up to a minute to find a match or it'll throw a connection error at you. This resembles For Glory Hell where Nintendo softbans systems and only banned users can play against each other. It is next to impossible to find For Fun matches which generally usually start with 3 people already in the room. I switched to the legacy version 7.1 and all my issues were resolved. Loading times decreased and I had next to no connection errors and was thrown into matches instantly and got full For Fun rooms.

**Steps to reproduce:**
The best way to compare these bugs is to either
1. Try to connect to a For Fun Smash room. Usually on normal systems you'll join a full room with 2-3 others already waiting. With Rosalina you're usually stuck with yourself.

2. Try to connect to a 1v1 For Glory room. On normal systems you can usually find people in a matter of 5-10 seconds between players. On Rosalina this resembles For Glory Hell in which a full minute can pass before you connect to someone else.

AuroraWright, 2017-07-08:
Do you use NTR?

SilentHeII, 2017-07-08:
I do not use NTR with Smash. I'm playing vanilla Smash.

AuroraWright, 2017-07-08:
So to be 100% sure NTR was not launched at all before launching Smash?

SilentHeII, 2017-07-08:
I should have also mentioned I'm not the only one who has experienced this issue. http://gbatemp.net/threads/super-smash-bros-is-having-problems-after-switching-to-b9s.476556/

And yes. I never ever launched NTR before Smash. I only use NTR for Fire Emblem.

Hikari-chin, 2017-07-08:
Have had someone test this and they couldn't reproduce the issues stated. Will do more research tomorrow.

Jan200101, 2017-07-09:
Can it be that you patch certain things into smash with Game Patching ?
I also encounter the long loading times when using mods for smash

Sonlen1414, 2017-07-09:
Tested with a o3ds, latest Luma Master AND Legacy Branches, put in a cart, updated the game, and went to connect online. About 10 seconds exact to connect to an online game with both Luma versions, which isn't a long wait time at all.

Hikari-chin, 2017-07-09:
Two independent parties tested and failed to reproduce the issues stated. Closing.

closed 2017-07-09 by Hikari-chin