LumaTeam/Luma3DS #712: [BUG] Game patching is broken for Sega 3d Fukkoku Archives (JP) in Luma 8.0

chashodges, 2017-07-08:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.1-557f205

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Having a similar issue as this bug - https://github.com/AuroraWright/Luma3DS/issues/545
with newest nightly Luma (as of 8th July '17) but with the Japanese release (Sega Fukkoku Archives 1). Same error message regarding SD card not being present as reported by user Epicbeardman in previous thread. I have created a locale patch manually and via the community homebrew solution. The game boots regardless of whether Luma is set up to load patches, but it seems impossible to shake the error message and thus saving is not possible in game.

Title ID is 0004000000154000 by the way.

Sonlen1414, 2017-07-09:
Tested the game with a USA SysNAND on latest Luma Nightly, with a locale.text "JPN JP" as the contents, and Luma Game Patching enabled, no complaining about a lack of SD being detected.

closed 2017-07-09 by Hikari-chin

chashodges, 2017-07-09:
I fixed it!

Basically, it wasn't working on my old Luma setup 7.x so I tried the region patching which I believe used to be faulty for this game. When I updated Luma to its most recent version I kept the locale document as US EN or EUR EN, when really all it needed was a clean JPN file as Luma now supports it fine.

Basically I didn't think the Luma version could have been a variable in the issue. Still, all sorted now!