LumaTeam/Luma3DS #714: Injected H&S app Shutdown Issue

Mike15678, 2017-07-09:
**System model:**
New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.3.0-36U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
8.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (4)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
 I seem not able to shutdown properly after I run the H&S app I injected with FBI.cia through GodMode9 (You can use any .cia smaller than the H&S app) and then close it and try to power off my 3ds.

**Steps to reproduce:**
(Everything below is assuming that you already have GodMode9 installed)

1. Boot your device while holding (Start) to launch the Luma3DS chainloader menu

2. Launch GodMode9 by pressing (A)

3. Navigate to [0:] SDCARD -> cias

4. Press (A) on your .cia to select it, then select “CIA image options…”, then select “Mount image to drive”

5. Press (A) on the .app file, then select “NCCH image options”, then select “Inject to H&S”

6. Press (A) to unlock SysNAND (lvl1) writing, then input the key combo given

7. Press (A) to continue

8. Press (B)

9, Press the "Start" button to reboot

10. Open up the H&S app (which is now FBI) and press "Start" or the home button to go back to the home menu. 

11. Hold the power button on your 3DS to reach the "sleep mode" reminder.

12. Keep holding the power button or tap the "Power Off" button.

13. Luma gives out an "ARM9 prefetch abort".

**Dump file:**

[Luma3DS Dump.zip](https://github.com/AuroraWright/Luma3DS/files/1133489/Luma3DS.Dump.zip)

**Note:** I know that that I shouldn't report this issue to Steveice10, because he said that this issue is not FBI-related, as stated here: https://github.com/Steveice10/FBI/issues/350. This may be a GodMode9 issue, but I first want to see if it's a Luma3DS issue.

Hikari-chin, 2017-07-09:
Confirmed this happens. Occurs with MultiUpdater injected, not an FBI issue. Will do more research.

Jan200101, 2017-07-09:
Apps injected into H&S linger around and become a "Zombie"
You can see that in the process list of Rosalina

TuxSH, 2017-07-09:
Stop using injections, it's a very hacky way to do things and that bug is none of my regard.

Latest commit fixes a bug with FBI + 3dsx, here's a build of it: 
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1134030/boot.zip)

closed 2017-07-09 by TuxSH