LumaTeam/Luma3DS #716: Battery charge issue

RS55, 2017-07-09:
Before installing any CFW to my 3DS (then RXtools), that is to say with OFW, buy a Nyko Power Pack for my O3DS and had an average of more than 8/10 hours per load as a minimum with everything off (wifi on, 3D, Lighting level 1, power saving on...), But I have noticed that from a time to here with the latest updates of Luma3DS (not to say exactly) and now with the rosalina menu I have confirmed that it no longer does the battery calculation as it originally did with OFW.

Last night I did the test waiting for it to reach the 01% and started to count the hours that lasted on, previously had played about 4 or 5 hours with everything to the maximum, then put super smash bros as it is more demanding for the system and leave fighting the CPU, in total the console hard between 3 and 4 hours more in 01% with the LED flashing.

So, I thought my battery was being damaged but it seems the opposite, maybe this also affects other users even using the original battery because long ago I read users asking if Luma3DS with A9LH consumed more battery.

samexner, 2017-07-09:
I have the same extended battery. I can test this tonight probably.

Hikari-chin, 2017-07-10:
Duplicate of other reports. This is pretty much impossible. See [here](https://en.wikipedia.org/wiki/Lithium-ion_battery#Degradation) and [here](https://en.wikipedia.org/wiki/Nocebo).

closed 2017-07-10 by Hikari-chin

RS55, 2017-07-10:
Well I doubt very much that it is degradation of my battery because it still works as the first day in terms of duration as I said before and believe me, I would not make a subject regarding this without knowing anything about how a Lithium-ion battery works, but good for me is no coincidence that the percentage is affected just when I pass to luma because otherwise the battery would NEVER last for almost 4 hours at 01% but well...

Hikari-chin, 2017-07-10:
Looking at that device, it's actually an extended battery that replaces the original back plate; given this, incorrect reporting isn't surprising. Many devices assume a certain capacity when doing battery life calculations; this issue also occurs on PS4 controllers, for example.