LumaTeam/Luma3DS #717: New3ds C-Stick Sensitivity Problem (InputRedirection)

Retro-Aisu, 2017-07-10:
I noticed while enabling the circle pad pro feature on MGS3/RE: Revelations, it takes you to a input test of the c-stick to test the sensitivity and when you move the analog stick with a external controller it sort of makes a "True" input. Even if you move the actual analog stick slightly it will register the c-stick as a full input. Now this doesn't matter for games like Smash but it does makes games where utilizing camera movement very weird.

(Tested this on a DS4 using Ds4 Windows, Gamecube Mayflash Adapter using x360ce, and an Ps3 Afterglow controller using x360ce.)

**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.1 

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: ( )

PIN lock: ( )

New 3DS CPU: (x)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Dump file:**

HugaShire, 2017-07-10:
This is due to the C-Stick sensitivity being high since it doesn't even move. It can be fixed by changing your controller sensitivity to really low. I could be wrong, I don't have an N3DS or a N2DS to test with.

Hikari-chin, 2017-07-10:
@HugaShire is very likely correct; the C-stick's sensitivity is very high as it's stationary; an analog stick with much greater freedom of movement would naturally cause a more drastic response on movement.

closed 2017-07-10 by Hikari-chin