LumaTeam/Luma3DS #720: Homebrew applications crash immediately after opening (luma 8.1)

Vaspro123, 2017-07-11:
System model:

New 3DS (XL)

SysNAND version (+emu/redNAND version if applicable):

11.2.0-35E SysNAND

Entrypoint (How/what you're using to boot Luma3DS):

Boot9Strap 1.2

Luma3DS version:

8.1 stable

Luma3DS configuration/options:

Default EmuNAND: ( 1 )

Screen brightness: ( 4 )

Splash: ( Off )

PIN lock: ( Off )

New 3DS CPU: ( Off )

Autoboot EmuNAND: ( Off )

Use EmuNAND FIRM if booting with R: ( Off )

Enable loading external FIRMs and modules: ( Off )

Enable game patching: ( Off )

Show NAND or user string in System Settings: ( On )

Show GBA boot screen in patched AGB_FIRM: ( Off )

Patch ARM9 access: ( Off )

Set developer UNITINFO: ( Off )

Explanation of the issue:

Attempting to use an application inside homebrew channel (.CIA) but when choosing one it goes to a crash dump screen, when attempting to use an exploit such as soundhax crashes immediately 

Steps to reproduce:

    Opening a homebrew application through the homebrew app (.CIA)

    Using an exploit such as soundhax to load homebrew

Dump file:
[image]

Dump files (0) through oppening application and (4) through using soundhax to load homebrew
[crash_dump_files.zip](https://github.com/AuroraWright/Luma3DS/files/1139593/crash_dump_files.zip)

fincs, 2017-07-11:
Please update your copy of [hbmenu](https://github.com/fincs/new-hbmenu/releases).

Vaspro123, 2017-07-11:
That has seemed to work thank you

closed 2017-07-11 by Vaspro123