LumaTeam/Luma3DS #721: Feature Request: Cripple the HTTP-sysmodule checks

rotomington, 2017-07-11:
Would it be possible for Luma to patch out the update checks, like ctr-httpwn? Luma already patches some sysmodules to allow old exploits to work on new firmwares, so why not re-introduce the httpwn exploit and have the exploit in RAM on boot? This would also allow O3DS users to play Extended RAM games on older firmwares for those who don't want to update (Due to some exploits not working or if future updates break Luma).

TuxSH, 2017-07-11:
First of all, there **NO** reason not to update.

> Would it be possible for Luma to patch out the update checks, like ctr-httpwn?

Yes, but there's no point. Also the checks are not done by the HTTP-sysmodule itself.

> Luma already patches some sysmodules to allow old exploits to work on new firmwares

It doesn't delibarately allow old exploits to work on old firmwares, except maybe for *hax on 11.3 and 11.4 w/ 11.2 payloads but that's all.

> so why not

Yeah, "just" do it. /s

> have the exploit in RAM on boot?

You have no idea of what you're talking about.

> This would also allow O3DS users to play Extended RAM games on older firmwares for those who don't want to update

They always were able to.

closed 2017-07-11 by TuxSH

rotomington, 2017-07-11:
Thanks for taking the time to reply, but I have a question

Although people on Luma are usually safe to update, what happens if an update is rolled out that breaks Luma or another part of the system (Like 11.3 and Ocarina of Time 3D)? Would this be considered useful then?

Hikari-chin, 2017-07-11:
It's very unlikely for any update to "break" Luma, but if that did occur, it would be fixed very quickly. There's really no point to this.

rotomington, 2017-07-11:
What about when Nintendo breaks their own stuff (Some DS games/DSiware do not work with 11.4 or the 11.3 and OOT thing)? That's the last excuse I can think of besides personal choice. 

Thanks for the answers so far though!

Margen67, 2017-07-11:
>Some DS games/DSiware do not work with 11.4

That wasn't Nintendo's fault. That was due to broken reboot patches, and was fixed.

The 11.3 OoT slowdown, however, is Nintendo's fault, and was fixed in 11.4.
Luma also had a patch that fixed the slowdown for 11.3 but that was removed since it broke other things on 11.3> and was outdated anyway.

@rotomington If you're worried about that just make NAND backups and use an emuNAND.

james-d-elliott, 2017-07-13:
I think this more belongs as a "game" patch (similar to SciresM's SSL patch) to the NIM module so GetSystemTitleHash always skips the actual check entirely. Don't really think it needs to be core Luma3DS functionality.