LumaTeam/Luma3DS #722: InputRedirection bug

Diego788, 2017-07-11:
**System model:**

Old3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

SysNAND 11.5.0-38U (bug happened even on 11.4)

**Entrypoint (How/what you're using to boot Luma3DS):**

SigHax (Boot9Strap)

**Luma3DS version:**

8.1 Stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: (X)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: (X)

--

**Explanation of the issue:**

first of all, i start InputRedirection in 3DS
then i start the app in my pc (InputRedirectionClient-Qt.exe) so i can use my joystick
i can control my 3ds with my joystick very good, every key works perfectly, but there's an issue
the only thing i can't do is use the Down-Left direction in circle pad, i can go left, right, down, up,        
up-right, up-left, down-right.... but the only direction that doesn't work is down-left

i tried with other joysticks but the issue continue, i tried using my keyboard (map the keyboard as  joystick and using it in InputRedirection) but even with what the down-left direction doesn't work... 
what can i do?

Hikari-chin, 2017-07-12:
Likely your computer or the client itself; did you try a different one? This would be a fairly major loss of functionality and hasn't been reported by anyone else, so I'm guessing it isn't a bug with Luma itself. I also asked a few users, and they didn't report having this issue.

closed 2017-07-12 by Hikari-chin

Diego788, 2017-07-14:
perhaps.. i'll try other .exe so i'll know if it was the program or my connection