LumaTeam/Luma3DS #723: Chainloader not working with Luma 8.1

PhoenixGamer, 2017-07-12:
**System model:**

[Old 3ds]

**SysNAND version (+emu/redNAND version if applicable):**

[11.4.0-37E Sysnand]

**Entrypoint:**

[Bs9]

**Luma3DS version:**

(8.1 Stable)

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (off)

PIN lock: (off)

--

Autoboot EmuNAND: (No Emunand)

Enable loading external FIRMs and modules: (yes)

Enable game patching: (yes)

Show NAND or user string in System Settings: (yes)

Show GBA boot screen in patched AGB_FIRM: (no)

Patch ARM9 access: (no)

Set developer UNITINFO: (no)

Disable ARM11 exception handlers: (no)

--

**Explanation of the issue:**
Luma won't start chainloader by presin start at boot

**Steps to reproduce:**

1.Put any chainload payload in plugin folder

2.start 3ds while holding start

**Dump file:**

No dump file

astronautlevel2, 2017-07-12:
Hello,

Can you send a screenshot of the payloads folder?

Shadow322, 2017-07-12:
its because u have a code.bin in rw/luma on nand as well as sd:/luma

ihaveamac, 2017-07-12:
@Shadow322 umm what? how is this related?

PhoenixGamer, 2017-07-12:
theres no code.bin in on my sd

astronautlevel2, 2017-07-12:
@PhoenixGamer 

A couple things to try:

1. Delete the Y_GodMode9 folder
2. Update the X_GodMode9.firm to the latest version on the GM9 repo
3. Rename it to just GodMode9.firm and hold start on boot

PhoenixGamer, 2017-07-12:
doesn`t work

Hikari-chin, 2017-07-12:
Can't reproduce. Also, if you've only got one payload in /luma/payloads holding Start with launch it, instead of the chainloader menu.

closed 2017-07-12 by Hikari-chin

HugaShire, 2017-07-12:
@Hikari-chin You should remove the "Can't Reproduce" label since there is only one payload on in there.

Hikari-chin, 2017-07-12:
@HugaShire I tested with both multiple payloads and a single payload in the /luma/payloads folder, and neither case had this issue, so indeed, I couldn't reproduce it.