LumaTeam/Luma3DS #724: Problems when I try to Update Luma3DS (7.0  to 8.1) by Luma3DS Updater

andrejones82, 2017-07-13:
I open the Luma3DS Updater and choose "Install normal version". I see that it gets to make the file download (Luma3DSv8.1.7z) but when it tries to extract it shows me an error message: 

Extracting payload [ERR]
FATAL: Could not find arm9loaderhax.binFATAL 
Could not get A9LH payload...

Can I do something to fix it?

ihaveamac, 2017-07-13:
lumaupdater is a separate program unrelated to Luma3DS.

if you are still on arm9loaderhax you would update to boot9strap first at https://3ds.guide/a9lh-to-b9s, which will set up v8.1 and a newer lumaupdater that supports newer luma versions.

Shadow322, 2017-07-13:
be sure to update to B9S v1.2 before putting Luma3DS v8.x on your 3ds https://3ds.guide/updating-b9s

Margen67, 2017-07-13:
>Can I do something to fix it?

Yes. You can learn how to read.

># Luma updaters that don't support Boot9Strap/Sighax won't work.
># This is due to support for non-B9S/Sighax entrypoints being dropped.

closed 2017-07-13 by Hikari-chin