LumaTeam/Luma3DS #726: New 3DS menu -> Clock rate setting not remembered on reboot, or power off and back on

AndrewHend, 2017-07-13:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-38U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap v1.2

**Luma3DS version:**

8.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: (off )

PIN lock: (off )

New 3DS CPU: (Clock + L2 )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

If one sets the 'Set Clock Rate to 804mhz', from the default 268mhz in the New 3DS menu of Rosalina and then saves the changes via Miscellaneous options 'save settings' also in the Rosalina menu, the cpu clock speed change does not stick and is always reverted back to 268mhz upon system reboot/power off + power back on.

It is expected that this settings change would stick in between reboots and power offs, as per set by the user.  All other settings changed stick if saved from within the Rosalina menu, so why not this?

**Steps to reproduce:**

1. Enter the Rosalina menu -> New 3DS menu -> and change the set clock rate option from 268mh to 804mhz.

2. Click B to go back and go into Miscellaneous options menu in Rosalina -> save settings.

3. Reboot, or power off and then back on your New 3DS system.  Go into the Rosalina menu -> New 3DS menu and notice that the clock rate is no longer set to 804mhz and has reverted back to 268mhz.

**Dump file:**

Hikari-chin, 2017-07-13:
There's an option in Luma's config menu (hold Select at boot) for this. Rosalina's "Save settings" is only for remembering the takeover title and key combination.

closed 2017-07-13 by Hikari-chin

AndrewHend, 2017-07-13:
Ok, I get this, thanks for sharing, however, I can't help but argue the inconsistency here.  If the setting is set to 804mhz in the Luma Config Menu, which comes before the Rosalina Menu is loaded, should the Rosalina Menu not read this setting correctly and display it accordingly?  If the Luma config menu is set to 804mhz, why would the Rosalina menu show 268mhz?  This is where the confusion lies.  If one sets it once, it should read that way everywhere, meaning 804mhz set in Luma Config menu should have Rosalia read this back to the end user in the same manner.  Why the difference?  This tells end users that the setting is 268mhz, as that is what is read, even if it isn't!

I appreciate your time none the less, thanks.

Hikari-chin, 2017-07-13:
That isn't the case. If you set your clock rate to force 804MHz mode in Luma's config, Rosalina will display it as such. Additionally, on N3DS the Home Menu runs at 804MHz by default. Rosalina refreshes the displayed clock rate as it changes (though, it didn't do this pre-8.1 due to a bug) and will display correctly non matter the scenario, forced 804MHz mode or not.

AndrewHend, 2017-07-13:
Hi again, speedy replies, thanks so much for this.

For me, what you mentioned above is not the case.  In the Luma config menu I have it set to Clock + L2, which I assume means the higher clock rate of 804mhz, plus the additional L2 cache being enabled??

Going into the Rosalina Menu, it is shows 268mhz.  Unless I am misreading what you wrote above, you said that forcing the clock rate in Luma Config, should have Rosalina reading the same and this is not happening, in fact, regardless, Rosalina is reading 268mhz.  I can change it to read 804mhz, but as soon as reboot the system, it shows 268mhz, which is why I started this report.

Just to clarify...

Is the Luma config menu of Clock + L2, what should be enabled to have 804mhz overclocked operation?

If this option is marked with an X, is Rosalina suppose to report back 804mhz then?

Sorry for the redundancy here, I am just getting differently then what I am reading above from your replies.

Hikari-chin, 2017-07-13:
Just retested this. As expected, Rosalina shows the correct clock rate regardless of if the option is forced or not. You're misreading it; Rosalina showing  "Set clock rate to 268MHz" would mean the system is currently in  804MHz mode, since by choosing that option you'd be [setting](http://www.dictionary.com/browse/set) it to 268MHz.

AndrewHend, 2017-07-13:
oh shit.  So sorry about all this, makes total sense now, I was just reading it backwards.  The menu is showing what you would actually change the clock rate to, not what it is actually set at now!  When I kept seeing 268, I thought it was showing me what it was set at currently and that I had to change this to 804mhz in order to take advantage of the faster speeds.

It appears having clock + l2 set in the Luma Config menu and a reading of 268mhz in the Rosalina menu, is exactly how it should be, if one wants 804mhz to be the set cpu speed.  I had it right the entire time, but misread it!

You are busy and I apologize for wasting so much of your time.  Thanks for all your patience with me, I really appreciate it and will leave this be now.

Keep up the great work, great service from you guys, considering for the most part, this project is being done for free.

Hikari-chin, 2017-07-13:
Don't worry about it; we all make mistakes sometimes.

AndrewHend, 2017-07-13:
You are most kind and understanding.

Do you guys take donations at all?  If so, where do I send them?  I am very new to this (as you can tell), but am already reaping the benefits of Luma immensely and am really appreciative of this software.  I don't have much to offer, but would like to help if I could.

AndrewHend, 2017-08-05:
I am sorry to drudge this back up again, but going into the menu and seeing the way things are currently listed, gave me an idea.  I would like to think I am not the only one to make the mistake above that I did, in misunderstanding/misreading, what was being relayed to me, in terms of what the CURRENT cpu speed was, vs what it was asking you to SET IT to.  

What if the following could take place?  Instead of it reading as follows:

Set clock rate to 268MHz, or

Set clock rate to 804MHz

It could read as:

Set clock rate to 268MHz (currently 804mhz), or

Set clock rate to 804MHz (currently 268mhz)

If I were to see the above, I would never have confused things the way I did, as it clearly tells me what the cpu is currently set at and what it will be set to, from the current setting.  I think this could help avoid confusion.

I know, you guys are busy and this is petty, but thought it might be worth mentioning, in case it is helpful to end users who may read it wrong.  It may not be that long/difficult to implement either, but what do I know?

Thanks for your time in reading this.