LumaTeam/Luma3DS #727: An exception occurred

mastergamerdragon, 2017-07-14:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

mastergamerdragon, 2017-07-14:
O3ds reciently programing in Boot9Strap 1.2 and luma 8.1 sysnand updated 11.5.0-38U fbi injected in h/s all working fine but i take shutdown console crash An exception occurred arm9 
[image] please help to fix this error thanks

mastergamerdragon, 2017-07-14:
I only marking this 
Enable game patching
Show NAND or user string in System Settings
Show GBA boot screen in patched AGB_FIRM

fox8091, 2017-07-14:
You didn't even fill out the template, wow. What did you do to make this happen? Is it reproducable? Did you even bother uploading the dump file?

Hikari-chin, 2017-07-14:
Duplicate of other issues. The solution stated in the others is:

**Stop using injections.** They're hacky and there's no reason to use them anymore.

closed 2017-07-14 by Hikari-chin