LumaTeam/Luma3DS #728: Crash when launcher hbmenu

iamliammckimm, 2017-07-15:
**System model:** old3ds
**SysNAND version:** Sys 11.5.0-38U
**Entrypoint:** b9s
**Luma3DS version:** 8.1

**Luma3DS configuration/options:**

-   Default EmuNAND: 1
-   Screen brightness: 4
-   Splash: Off
-   PIN lock: Off
-   Autoboot EmuNAND: False
-   Use EmuNAND FIRM if booting with R: False
-   Enable loading external FIRMs and modules: False
-   Enable game patching: False
-   Show NAND or user string in System Settings: True
-   Show GBA boot screen in patched AGB_FIRM: False
-   Patch ARM9 access: False
-   Set developer UNITINFO:False
-   Disable ARM11 exception handlers: False

**Explanation of the issue:**

I boot up my 3ds and try to open the hbmenu with the hblauncher_loader cia and it gives me a crash message:

[image]

**Steps to reproduce:**

1. Launch hbmenu with hblauncher_loader cia 

**Dump file:**

[crash_dump_00000000.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/1150048/crash_dump_00000000.dmp.zip)

Hikari-chin, 2017-07-15:
Update your [boot.3dsx](https://github.com/fincs/new-hbmenu/releases).

closed 2017-07-15 by Hikari-chin

iamliammckimm, 2017-07-15:
i had but idk why it works now :/ sorry