LumaTeam/Luma3DS #729: [3DS Game Crashing] Monster Hunter 4 Ultimate (digital version from the eshop)

ThoDM, 2017-07-15:
System model:

2DS

SysNAND version (+emu/redNAND version if applicable):

11.4.0-37E SysNAND

Entrypoint (How/what you're using to boot Luma3DS):

Boot9Strap

Luma3DS version:

8.1 stable

Luma3DS configuration/options:

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

Explanation of the issue:

I'll repeatedly get ARM11 exception handler messages with the game freezing and the handler generating a dump. It happens out of nowhere and can go hours of playing without getting one crash or something I do will crash it right away, repeatedly and consistently (we will use that for the reproduction). Certain locations in the game are more prone to crashing than others and it has never crashed during a quest so far. It's much more prone to happen in Val Habar for example (can't spend more than a minute going around talking to wycoon and stuff without it crashing) while I've never had a problem in areas like the Elder Hall (but can still intentionally cause it to crash with a certain couple of steps). Game is an official eshop version I've had for years and never had a problem with it before installing Boot9Strap and Luma 8.0. I'm not running any additional apps/utilities or that sort of thing with the game (eg: NTR), just playing normally.

Steps to reproduce:

Method 1- Go around Val Habar, sticking close to The Man (smith) usually doesn't cause it, so talk to Wycoon, eat, shot, use box, game should crash after about a minute. Happens more often after taking every item you had put in for multiplication by the Wycoon.

Method 2- The most consistent one I've found is one involving the Combo List when accessed from the in-game menu (the one with the Start button, box combo list doesn't trigger this). I have these exact items in my pouch:  Book of Combos 1-5, Potion x10, Mega Potion x10, Mega Dash Juice x5, Disposable Earplugs x3, Powercharm, Powertalon, Armorcharm, Armortalon, Well-done Steak x10, Paintball x10, Tranq Bomb x8, Net x10, Trap Tool x2, Pitfall Trap x1, Shock Trap x1, Whetstone x20. The moment I select the Combo List option in the menu the game will crash and the ARM 11 exception handler message will pop-up.

Dump file:

[crash_dump_00000010.zip](https://github.com/AuroraWright/Luma3DS/files/1150243/crash_dump_00000010.zip)

If needed I can repeat the methods and trigger another crash for more dumps. This problem is really annoying when it happens during online play since most interesting halls get full really quickly and having my 2DS shut off on me makes things considerably hard. Any working solution is highly appreciated.

Hikari-chin, 2017-07-16:
Couldn't reproduce this. You've likely got a bad SD card or a bad install; try "Repair software" from the eShop listing.

closed 2017-07-16 by Hikari-chin

ThoDM, 2017-07-16:
Already have tried with two different SD cards (main one 128GB and secondary I had before cfw 32GB) and tried repairing too. Also, I went back to 7.1 version of Luma to test and everything works fine, but I need homebrews on my 2DS so I somehow need to make it work with the 8.1. Should I try a nightly and see if it fixes? When on 8.0 the game hardly ever crashed btw. Currently have left it on 8.0.

Hikari-chin, 2017-07-16:
Nothing changed between 7.1, 8.0, and 8.1 that could cause these effects. Additionally, since I couldn't reproduce it (also had someone else test), I'm chalking the "differences" between versions to be random chance or the nocebo effect. Whatever the case, this doesn't seem to be a Luma issue.