LumaTeam/Luma3DS #730: [Request] Reboot into CTR-mode titles with TWL_FIRM

Robz8, 2017-07-17:
So @ApacheThunder figured out how to reboot into DSiWare titles in TWL_FIRM, by providing a full title TID at RAM area 0x02000310 (e.g. 000300044B513945, but appears in RAM as 4539514B04000300), and something else at 0x02000000 (check attached file).

He thought of the idea of letting NS check a byte in that RAM area (0x02000316), so if it's 03, it's a DSiWare title, but, if it's 04, then it's a CTR/3DS mode title. If one provides a CTR-mode title's UniqueID, then a CTR-mode app will boot.

Since many people want to return to TWLoader when running DS games, this idea should solve this issue.

[sr_data_fourswords.zip](https://github.com/AuroraWright/Luma3DS/files/1151590/sr_data_fourswords.zip)

ApacheThunder, 2017-07-17:
Actually mostly shutterbug's discovery FYI. Only the suggestion for changing behavior of NS came from me. :P

Robz8, 2017-07-18:
We found out that we do not need any data at 0x02000000 for soft-reset.
We only need to provide a full title TID at RAM area 0x02000310, and also a CRC16 value (byteswapped) at 0x02000306.
The CRC covers 0x18 bytes, 0x02000308 to 0x0200031F.

CRC settings:
Polynomial: 8005
Initial value: FFFF
Output XOR: 0
Reflection:
Input: Yes
Output: Yes

ApacheThunder, 2017-07-18:
Note that those ram offsets are reletive to the ram setup during legacy mode. ;)

jupitteer, 2017-07-21:
This will be great if it is implemented. Best of luck.

Robz8, 2017-08-12:
Since boot9strap can now load bootonce.firm on boot, this is now closed, as I have a new idea.
I'll open another issue about it.

commit 44758715 (2017-07-18): Log "logged" errdisp messages in /luma/errdisp.txt, fixes #730
closed 2017-08-12 by Robz8