LumaTeam/Luma3DS #731: [Bug] Screenshot feature does not reflect color values given by Power Saving Mode

NinjaBoyLao, 2017-07-17:
**System model:**

Old 3DS XL, TN screens

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-38U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9strap

**Luma3DS version:**

8.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (no)

Screen brightness: (3)

Splash: (no )

PIN lock: (no )

New 3DS CPU: (no)

--

Autoboot EmuNAND: (no)

Use EmuNAND FIRM if booting with R: (no)

Enable loading external FIRMs and modules: (no)

Enable game patching: (yes)

Show NAND or user string in System Settings: (no)

Show GBA boot screen in patched AGB_FIRM: (yes)

Patch ARM9 access: (no)

Set developer UNITINFO: (no)

Disable ARM11 exception handlers: (no)

--

**Explanation of the issue:**
Taking a screenshot while power saving mode is on results in a screenshot as if power saving mode were not on.

**Steps to reproduce:**

1. Turn on power saving mode

2. Take a screenshot of something blue [maybe the FBI icon](https://github.com/Steveice10/FBI/issues/368)

**Dump file:**
no crash

NinjaBoyLao, 2017-07-17:
@Margen67 what's up? Is something unclear? There are screenshot files and an actual camera image reflecting this in the link provided

Margen67, 2017-07-17:
@NinjaBoyLao https://youtu.be/no0qB8CVspY

Power saving mode dims the backlight. If you only make the screen look dimmer (which you can do with an NTR overlay) it doesn't save any battery at all. The closest to that is OLED which can save power by turning off an OLED if it's pure black, but the 3DS doesn't have an OLED display.

NinjaBoyLao, 2017-07-17:
Power saving mode dims the backlight of the display, and compensates for this by increasing the gamma, or white-point. It is a software function, not a hardware one, and image capturing programs should be able to detect it. The screen on my O3DS XL is a TN LCD. When active, the backlight is dimmer, reducing power usage, but the actual colors of the screen change slightly to be more or less contrasting calculated by the original color (whites and blacks both become "warmer" variations to compensate, so that the change in brightness isn't as noticeable). There was a crappy explanation of this given in an old episode of *Iwata Asks* 

The built-in screenshot function on the 3DS reflects this, and an image manipulation program like GIMP can be used to check the hex color values for the exact same pixels of a power saving screenshot vs a regular one. 

I'm not sure what you're talking about for the overlay, but if an NTR overlay actually does dim the brightness, even without adjusting the colors it should still save battery life. The gains from using power saving mode on the highest brightness settings are somewhere between 10-20%, so I'd expect slightly less from just an overlay.

Hikari-chin, 2017-07-17:
I think preserving color accuracy when taking a screenshot would be considered a feature, not a bug.

closed 2017-07-17 by Hikari-chin

NinjaBoyLao, 2017-07-17:
@Hikari-chin i edited the title to reflect the nature of the issue

Gudf, 2017-07-18:
You should put this in #619