LumaTeam/Luma3DS #732: Failed to apply 1 FIRM patch(es) when boot DevFW on Retail o3DS XL

NgDTien, 2017-07-18:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-38E SysNAND, 0.0.0-00 EmuNAND(DevFW)

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

8.1 stable 

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (0)

PIN lock: (0)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: (X)

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: (X)

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Failed to apply 1 FIRM patch(es) when boot EmuNAND(DevFW)

**Steps to reproduce:**

1.

2.

**Dump file:**

astronautlevel2, 2017-07-18:
`0.0.0-00 EmuNAND(DevFW)`

This makes no sense, the DevFW has version numbers just like the retail FW.

If you mean Factory Firmware, that's pre 1.0, and there's no way that'll work with Luma, for similar reasons that 1.0 and 2.0 don't work. Besides, no one has all the sysmodules for factory firmware - you can't get that booting at all.

closed 2017-07-18 by Hikari-chin

HugaShire, 2017-07-19:
@astronautlevel2 It actually doesn't most of the time. Devfirmwares have 0.0.0-0 as stated [here](https://gbatemp.net/threads/tutorial-installing-dev-firmware-and-dev-apps-on-retail-3ds.429773/)

astronautlevel2, 2017-07-19:
That only refers to what version is displayed in CVER, not the actual firmware version.

HugaShire, 2017-07-19:
Happens to me too.

ihaveamac, 2017-07-19:
dev firm does have a version, it's just not reflected in cver/nver.

I tried booting 2.51 dev nfirm on a old 3DS retail unit and it booted to home menu. emunand was not used.

HugaShire, 2017-07-19:
@ihaveamac Even when I did it to sysnand it gave the error. For some odd reason the Boot9Stap_dev.firm won't install "says not Sighaxed". Are you using the devkit? Or is it because from that iso site the update is outdated

Hikari-chin, 2017-07-19:
ihaveamac tested on a retail device. You're probably trying to boot an external firm whose version is too low. However, the Luma issues aren't for tech support, try the NH Discord.

HugaShire, 2017-07-19:
@Hikari-chin, Sorry about. Got it working. Thanks.

michelemassa5599, 2022-04-09:
That has been projected usable file found..;)