LumaTeam/Luma3DS #733: rosalina hb redirection freezes

YamazakiRobert, 2017-07-19:
**System model:**
New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0.38U SYSNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

boot9strap

**Luma3DS version:**

8.1 stable or 8.1 nightly 7e7ab12 2017-07-18

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

i have switched the health app & safety information with hb launcher and saved the options 
i only have the boot.3dsx the 2.0 version
and on 3ds folder
- fbi.3dsx v2.8

i can navigate and close the hb but when i load the fbi.3dsx or any other 3dsx the new 3ds freezes

**Steps to reproduce:**

1. replace the health app & safety information with hb launcher and saved the options 

2. load the fbi.3dsx or any other 3dsx 

**Dump file:**

The  system only freeze and dont get to the dump screen
no dumps on luma folder

IngeniousDefault, 2017-07-19:
#679

Hikari-chin, 2017-07-19:
Known issue. Being looked into. Don't use H&S as your takeover app for now.

closed 2017-07-19 by Hikari-chin