LumaTeam/Luma3DS #735: [Feature Request] Plugin Support

DevShadd, 2017-07-19:
Plugins are also needed more and more for the 3DS mod scene. Having plugins built into Luma would allow simpler ways to setup and use plugins. Plugin support would also allow further improvements to overall system stability and other minor adjustments to enhance the user experience. ( ͡° ͜ʖ ͡° )

Plugins should have the following support:
►Plugins can be enabled/disabled via Rosalina menu
►Plugins should be able to not run until under certain circumstances when a plugin is enabled (when an app or game loads, certain system settings are enabled, etc.)
►Allow plugins to force disable manual enabling/disabling to the plugin and other certain plugins (to prevent any issues)

Margen67, 2017-07-19:
#619

HugaShire, 2017-07-19:
Close this issue this is:

1. A duplicate 

2. Should belong in #619

closed 2017-07-19 by Hikari-chin