LumaTeam/Luma3DS #737: Updating from 7.1-a187c0c7 to latest hourly (3907c46) softbricks

supercarotte, 2017-07-20:
**System model** : New 3DSXL

**SysNAND version** : 11.5-E

**Entrypoint** : B9S

**Luma3DS version** : see title

**Luma3DS configuration/options**: 
Default EmuNAND: (1 )
Screen brightness: (4 )
Splash: (Off )
PIN lock: (Off )
New 3DS CPU: (Clock+L2 )
--
Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Enable game patching: (X )
Show NAND or user string in System Settings: (X )
Show GBA boot screen in patched AGB_FIRM: (X )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )
Disable ARM11 exception handlers: ( )
--

**Explanation of the issue:**
I simply booted Luma3DS updater on a spare device (which has B9S+7.1 Luma)  I was refreshing up and used it to update Luma to the latest hourly. After downloading and replacing boot.firm, my console shuts off.
When I press Power again, the blue led lights up for maybe half a second then switches off. Nothing else lights up (no screen, no other leds) and Start/Select don't trigger anything.

Reverting back to the old .firm fixes the issue, repeating the process leads to the same brick and "Enable exception handlers" option doesn't create any crash dump.

**Steps to reproduce:**
See above

**Dump file:**
Here are the 2 boot.firm files as well as lumaupdater logs since I couldn't get a crash log.
[Attached.zip](https://github.com/AuroraWright/Luma3DS/files/1161627/Attached.zip)
Boot.firm = 8.1-3907c46
Boot.firm.bak = 7.1-a187c0c7

vaguerant, 2017-07-20:
Sounds like you most likely need to update your boot9strap install, what version are you on? Current version is 1.2, from June 5, 2017.

astronautlevel2, 2017-07-20:
This isn't a bug - Luma 8.0+ is only compatible with boot9strap 1.2, whereas 7.1 is only compatible with boot9strap 1.0. https://3ds.guide/updating-b9s

This isn't a support forum - if you have issues, please take them to the Nintendo Homebrew discord.

closed 2017-07-20 by Hikari-chin