LumaTeam/Luma3DS #738: Crash launching system settings

MrDudeldo, 2017-07-20:
**System model:**

New 3DS XL (EU)

**SysNAND version (+emu/redNAND version if applicable):**

11.2.0-35E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

I started with a completely normal and stock N3DS XL - EU Version.
I followed the simple guide on 3ds.guide using all the lastest releases and everything worked and works fine but since I installed boot9strap and Luma3DS the 3DS crashes whenever I try to enter System Settings. Another error occurs when I enter the camera app (don't know if they are related or not)

launch System Settings crash: http://i.imgur.com/B9L9zDf.jpg
launch Camera Crash: http://i.imgur.com/0ZtyWH1.jpg

**Dump file:**

[crash_dump_00000013.zip](https://github.com/AuroraWright/Luma3DS/files/1163890/crash_dump_00000013.zip)

Hikari-chin, 2017-07-20:
Can't reproduce, unlikely to be a Luma issue.

closed 2017-07-20 by Hikari-chin

HugaShire, 2017-07-21:
Your apps are corrupted. Do a CTRtransfer, do a system update, or use Sysupdater to install 11.2

MrDudeldo, 2017-07-22:
> Your apps are corrupted. Do a CTRtransfer, do a system update, or use Sysupdater to install 11.2

Thanks for that, got into system update menu by pressing L+R+Up+A . System Settings works again after the update. Just the camera still doesn't function, but that's negligible I guess..