LumaTeam/Luma3DS #740: I cant access my home menu on my O3DS XL

jjja909, 2017-07-21:
Whenever I turn on my 3DS, an error screen popped up, saying  (in red text) "An error occurred (ErrDisp)
(White text) Error type: logged
Process ID : 15
Process title ID: 0x0004003000008f02

Address: 0x00164654
Error code: 0xd900458b

Press any button to reboot"
When I reboot, the error screen pops back up, Im on version 11.3, Luma 7.1. Somebody on the discord showed me a boot.firm that might help but it gave me a new error screen saying:
"(Red Text) An exception occurred
(White Text) Processor: ARM11 (core 0)
Exception type: prefetch abort (svoBreak)
Current process: menu (0004003000008F02)
R0 00000000 R1 00000000
R2 00000000 R3 00000001
R4 0FFFFF10  R5 D900458B
R6 00000000 R7 0FFFFFF2C
R8 00000000 R9 0032D848
R10 00343FAC R11 00000000
R12 00000000 SP 0FFFFEF0
LR 00164658 PC 00136F28
CPSR 60000010 FPEXC 400000700
You can find a dump in the following file: dumps/arm22/crash_dump_00000004.dmp
Press any button to shutdown"
(Stack dump)
Then I showed it to the discord and they gave me some other stuff and a new boot.firm, When I launch the 3ds now, the yellow light is on but not the screen, but when I turn it off, The confirmation screen pops up (The screen that confirms if you want to turn off the device) Then I pressed Home And the screen that says Home Menu with the white background and just freezes, Ive tried a ctr transfer, Ive cleared my home menu ext data, Ive tried a lot, Also A couple of people on the discord are having the same problem, Mine happened when I was playing pokemon sun, I went to the home screen with the app still open, I closed my 3DS and put it in my pocket, I took it out of my pocket and it was off! I turned it on and, error screen. Pokemon sun was downloaded on my 3ds and the thing that all the people with the problem had in common, was that we all had a ds game in the slot... Im not sure how to fix this please help.

Hikari-chin, 2017-07-21:
The behavior you describe sounds like a bad NAND or SD card rather than an issue with Luma; try restoring your NAND and/or checking your SD card with something like h2testw. 

If you continue to experience issues, try the [Nintendo Homebrew Discord](https://discord.gg/C29hYvh) for further support; the Luma issue tracker isn't the proper place for it.

closed 2017-07-21 by Hikari-chin

200zero, 2017-07-21:
we been trying Discord and no luck at all

HugaShire, 2017-07-21:
Try booting without a Game.

HugaShire, 2017-07-21:
Try booting with your SD card out, and then reinserting it after booting
Power off your device
Remove your SD card from your device
Power on your device
When the home menu appears, reinsert your SD card into your device
If this worked, you should clear Home Menu’s extdata by navigating to the following folder on your SD card: /Nintendo 3DS/(32 Character ID)/(32 Character ID)/extdata/00000000/
EUR Region: Delete 00000098
JPN Region: Delete 00000082
USA Region: Delete 0000008f
CHN Region: Delete 000000A1
KOR Region: Delete 000000A9
TWN Region: Delete 000000B1
Try booting without any cartridges inserted (including flashcarts)
If you have a hardmod and a NAND backup, flash the backup back to SysNAND
Try booting into recovery mode and updating your system
Power off your device
Hold (L) + (R) + (A) + (Up)
Power on your device
If you enter safe mode, update your device
Your device may be bricked. For support, ask for help at Nintendo Homebrew on Discord

RubberBrony, 2017-07-23:
Also, if you have a DS card with an invalid header or poor contacts, the HOME Menu will sometimes fail to load with no fatal errors reported. Remove the cartridge from the cartridge slot.