LumaTeam/Luma3DS #741: This GBA injected cias crashes are driving me crazy >.<

MG08, 2017-07-22:
**System model:**

N3DSXL

SysNAND 11.5.0 U

Boot9Strap/Sighax 

**Luma3DS version:**

8.1 stable 

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( x)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( x)

Show GBA boot screen in patched AGB_FIRM: ( x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
.
A lot of times when I boot a GBA injected game the console crashes and shows that "An error has occurred..." black error screen. It's been like that since I setup the console months ago, at that time I had A9LH.
My MicroSD is 128GB, and I have around 5GB free only. Cluster size WAS 64kb, y change it to 32kb but the error still persist, sometimes games boot ok, sometimes they don't, like 50%-50% -.-

**Steps to reproduce:**

1. Boot up a Gba injected cia game.

2. Wait 

3. Boot other if it didn't triggered the error.

4. ??????

**Dump file:**
[arm11.zip](https://github.com/AuroraWright/Luma3DS/files/1167135/arm11.zip)

Shadow322, 2017-07-22:
Disable N3DS Cpu clock speeds

MG08, 2017-07-22:
I tried it.
Boot a game, ok.
Boot a second one, ok.
Boot a third one, error screen...

Shadow322, 2017-07-22:
Then glad it's solved

Hikari-chin, 2017-07-22:
Can't reproduce. GBA VC games work fine for me.

closed 2017-07-22 by Hikari-chin