LumaTeam/Luma3DS #742: Rosalina New 3ds Menu Settings does nothing

bronozoj, 2017-07-23:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (1)

Splash: ( off)

PIN lock: (off )

New 3DS CPU: ( clock + l2)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: ( x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
tried tinkering with the rosalina menu and found out that it does not modify clock speed or l2

**Steps to reproduce:**

1. setting 3ds cpu in boot config to clock + l2 and modifying it in rosalina to remove l2 cache does not do anything as sonic and all stars racing transformed hardlock (as if l2 cache is enabled)

2. doing the opposite, setting new 3ds cpu in boot config to clock only and modifying rosalina settings to include l2 cache seems to not cause hardlock in same game

**Dump file:**

bronozoj, 2017-07-23:
forgot to add other info:
New 3ds XL on boot9strap 1.2
Luna 3ds v8.1 stable
sysnand 11.4.0-37U

HugaShire, 2017-07-23:
This is because you have the Luma Config make the speed of the CPU change already. The Rosalina menu won't do anything if the Luma Cofig has it.

Hikari-chin, 2017-07-23:
Works for me. Performance difference is noticeable in certain games, etc.

closed 2017-07-23 by Hikari-chin